pihomeserver/Kupiki-Hotspot-Script #88: bug

reigelgallarde, 2018-03-06:
SECRETKEY for freeradius and chilli are not being used in the script...

pihomeserver, 2018-03-06:
Thanks for the issue. I will update the script and set keys to random values

pihomeserver, 2018-03-06:
Corrected in the v1.8.2 (just published)

closed 2018-03-06 by pihomeserver

reigelgallarde, 2018-03-07:
why will JS library not work when the coova chilli key is not blank?

reading the code on ChilliLibrary.js both CHAP and PAP logon uses UAM secret.

So I'm curios because I'm using my own version of portal and with UAM secret it works.
or maybe because I'm not using the user/password field...

I'm using `logon2` which is the next step of `logon`.
May portal is related to what I've posted before here https://github.com/Kupiki/Kupiki-Hotspot-Portal/issues/6

`logon2` requires `username` and `response` parameters. This is good because `response` can only be used once.
server side you can use `chilli_response <challenge> <uamsecret> <password>` to get the `response`.
pass `response` back to the client side and use `logon2` to login the device.

With this idea by the way, you can create a social authentication ( related post https://github.com/Kupiki/Kupiki-Hotspot-Portal/issues/2 ).

reigelgallarde, 2018-03-07:
ahh.. I forgot I scratch using `logon2`... because on `logonStep2` it requires the password in a temp variable.
I don't want password to be visible on the client side because it can be viewed using debug mode on any browser.

What I end up doing was after getting the `response`, I created a url like this:
`var logonUrl = chilliController.urlRoot() + 'logon?username=' + escape(username) + '&response='  + response;`

visit `logonUrl` via ajax, the response is the same with `<IP>:<port>/json/status`.
or you can do `chilliController.refresh()` to check the `clientState`