pihomeserver/Kupiki-Hotspot-Script #91: Ip Range more than the one of the Hotspot

xerxes09, 2018-03-17:
Hey,

When I'm locked in in the WLan  of the raspberry, I can open all other IPs of the whole Ethernet. 
I can open the daloradius login page, the router, everything...

pihomeserver, 2018-03-18:
Hello
I guess it's not locked but log ? Which version of OS and of the script ? There is an iptable rule that blocks that

reigelgallarde, 2018-04-13:
I'm having same problem. But my knowledge on iptables is limited.
The router/modem itself where the pi connected is accessible... prone to hacking the router/modem settings.

pihomeserver, 2018-04-15:
Have to say that's a little bit more complex than expected. I'm able to block access to the IP of the raspberry but not the complete LAN ...

pihomeserver, 2018-04-15:
Ok i created a new branch called v1.8.7. Can you please give a try to it ?
If you don't want to install it without restarting from scratch you have to do following actions in `/etc/chilli/up.sh` : 
- Delete all lines at the end of the file 
```
LOCAL_IP=\`ifconfig \$HS_WANIF | grep -Eo 'inet (addr:)?([0-9]*\.){3}[0-9]*' | grep -Eo '([0-9]*\.){3}[0-9]*' | grep -v '127.0.0.1'\`
ipt -C INPUT -i \$TUNTAP -d \$LOCAL_IP -j DROP
if [ \$? -ne 0 ]
then
    ipt -A INPUT -i \$TUNTAP -d \$LOCAL_IP -p tcp -m tcp --dport $HOTSPOT_PORT -j ACCEPT
    ipt -A INPUT -i \$TUNTAP -d \$LOCAL_IP -j DROP
fi
```
- Add following lines at the end of the same file 
```
LOCAL_IP=`ifconfig $HS_WANIF | grep -Eo 'inet (addr:)?([0-9]*\.){3}[0-9]*' | grep -Eo '([0-9]*\.){3}[0-9]*' | grep -v '127.0.0.1'`
LOCAL_LAN=`ip -o -f inet addr show | grep $HS_WANIF | awk '/scope global/ {print $4}'`
ipt -I FORWARD 1 -i $TUNTAP -d $LOCAL_LAN -j REJECT
ipt -I INPUT 1 -i $TUNTAP -d $LOCAL_IP -j REJECT
```

I made some tests and seems to work as expected. 
**One point to confirm : do you prefer DROP or REJECT ?**
- DROP will make requests end after a long time as if the IP or service called not exist
- REJECT will make the requests end immediately saying that the connection is impossible.

I would prefer the first. Less user friendly but in my opinion more secure in order to avoid network discover

pihomeserver, 2018-04-17:
Merged with a REJECT iptables. Will be updated in the next release for a DROP iptables rule

closed 2018-04-17 by pihomeserver