pihomeserver/Kupiki-Hotspot-Script #92: Unable to extend signal coverage using wifi repeaters and extenders.

droecoleman, 2018-03-18:


pihomeserver, 2018-03-18:
Can you please provide more details ? I'm not an expert too i don't see why it's not possible and if it's related to the configuration of the hotspot or to the Raspberry hardware

droecoleman, 2018-03-19:
i was trying to extend the wifi signal of the pi by adding a repeater. However, when i try to connect using my phone or laptop to that repeater it won't let me connect and the status is always obtaining IP address. I did try another brand of repeater this time i can connect but no internet , the portal page wont come up. If i go and connect directly to pi built in wifi signal everything is fine.

pihomeserver, 2018-03-19:
Can you try to use the same repeater on another hotspot to check the issue is from the Pi and not the repeater ?
I will look what could be the reason of that issue

droecoleman, 2018-03-19:
yes, i have tried two different repeater on another hotspot(not Kupiki) and it works fine. However when i try this to Kupiki with repeater it wont relay anything.

pihomeserver, 2018-03-19:
If you installed the Kupiki-Admin interface, use it to change the channel and perform a new test, else update the /etc/hostapd/hostapd.conf to change the channel value by 0 to force the best channel with least interferences
```
channel=0
```
Don't forget to restart the hostapd service
If it doesn't work, try to add following options : 
```
ieee80211n=1
wmm_enabled=1
```

droecoleman, 2018-03-19:
how can i access Kupiki-Admin interface after installation? I have tested changing channel value to  0, it will cause the wireless signal to become off, so i assigned another value and the wifi name appear again.  ieee80211n=1
wmm_enabled=1 the problem still persist.

pihomeserver, 2018-03-19:
For the admin interface all is here : https://github.com/Kupiki/Kupiki-Hotspot-Admin-Install

Seems to set a particular configuration for hostapd for repeaters but don't know which one yet

droecoleman, 2018-03-19:
which address shall i go to get into Kupiki_admin after installation?

pihomeserver, 2018-03-19:
http://<same ip than daloradius>:3000
login/passwd : admin/admin
remember that it's still alpha, even some translations are missing in english, only french is fully available for now. 
The interface will not do more than updating the config file like you did manually
(of course any help to improve is welcome)

reigelgallarde, 2018-03-20:
I think wifi repeater will not work...
coovachilli works with IP... the repeater will have different IP range.

The only way for it to work is on bridge mode.. but then wifi on the pi will not be used.

This is my assumption, I will try when I get a hand on one wifi repeater.

droecoleman, 2018-03-20:
i added the following to hostapd.conf 
wpa_pairwise=TKIP
rsn_pairwise=CCMP. 
the repeater  now able to connect to the pi wireless signal. And can relay the captive portal for 6 seconds to client device after 6 seconds a warning  "internet unavailable".

pihomeserver, 2018-03-20:
Good news even if i don't understand why. WPA is disabled to make an open hotspot without any password request when joining the SSID

So to summarize, you see the portal, can connect but can't get an Internet access ?

droecoleman, 2018-03-20:
yes, and i cant get any ip address after 6 seconds from the  pi.

pihomeserver, 2018-03-20:
Can you try to stop freeradius and coova services then start them manually in debug mode to look at the generated messages ?

heneraloysat, 2018-10-12:
Hostapd is just for built-in wifi or wifi usb adapter

pihomeserver, 2018-10-12:
Default configuration is for the built in wifi. If you know the driver for the usb adaptater then replace the parameter in the script with the correct value before running the installation

reigelgallarde, 2019-01-09:
@droecoleman have you had success with this?
I have tried it now, but the problem is that those devices connected to the repeater are having the same mac address, which is the mac address of the repeater... a problem because all connected to the repeater share the same session...

droecoleman, 2019-04-18:
@reigelgallarde my apology for the late response, do you have some progress with this issue so far?. i temporarily stop my experimentation for this particular issue. But i will be back on track once i am done with the issue of batch add users not recognized by the portal even if it was created successfully.

reigelgallarde, 2019-04-18:
@droecoleman 
Not sure if acceptable, BUT I actually move on with another solution.. instead of extending the wifi of raspberry pi, I used a router that can be used as AP.. there are a lot.. and also has greater area coverage...

using USB to LAN, I added an AP... "wavlink n300" is an example that I used... 
I also used some script I found on the internet that forward internet on eth1

so instead of wlan0 on the script, you will have to use eth1 now...

droecoleman, 2019-04-18:
@reigelgallarde oh, that is good to know that. I will try that one of these days. I am really stuck with the batch add users problem. I hope you can send me your solutions on it.

reigelgallarde, 2019-04-18:
[image]

here's a sample picture of my setup...

but I'm really on a different level on this script...
you can check at my website: [https://mavispisowifisolutions.com/piso-wifi/](https://mavispisowifisolutions.com/piso-wifi/)
You can also look at some [videos here](https://www.facebook.com/pg/MavisPisoWifiSolutions/videos/) to see what I mean about different level..

If I can get a time, I will try to add my solution as a PR here...

The script to use eth1 as AP, I got from here https://github.com/oblique/create_ap/issues/61

droecoleman, 2019-04-25:
@reigelgallarde i am not successful of using the script for the AP. can you please elaborate on how to do it? i am struggling.

closed 2019-04-26 by droecoleman