pihomeserver/Kupiki-Hotspot-Script #93: Batch add users

droecoleman, 2018-03-20:
The lists of users that were added through "Batch add users" are not recognized in portal users log in.

pihomeserver, 2018-03-20:
Can you please share some records of the radcheck table of the radius database ?

droecoleman, 2018-07-30:
i am always getting json request time out. even after i have reinstall several times, how can i make this to work?

pihomeserver, 2018-07-30:
It's due to your configuration. Please open a new issue, follow the requirements to help people to investiguate
Also search existing closed issues. Same problem as been raised and was due to ip settings

droecoleman, 2019-04-18:
@reigelgallarde, do you have some recommendation on how to solve this? i am really stuck on this problem. Please help.

pihomeserver, 2019-04-18:
As said if you don't provide any additional details except "it does not work" how can we help ?
If you get the timeout that's your configuration the issue origin not daloradius. The timeout is related to the communication between the portal in the client browser to the hotspot server. 
You should start on that.

reigelgallarde, 2019-04-18:
@droecoleman 
I can't give solution because I can't reproduce the problem... also "Batch add users" is just a repeat process of  "add user", so there should be nothing wrong with that...

droecoleman, 2019-04-18:
i also thought that so. But if you have time please try. and send me your results. T.Y

pihomeserver, 2019-04-18:
Does it work for one user ?

droecoleman, 2019-04-18:
for new user or one user it does work. but in batch add users wont work, passwords and users will be created successfully however when used it doesnt work. invalid username and password.

pihomeserver, 2019-04-18:
My opinion : because in batch mode the attribute password type is not the same than manual creation and it does not match the Freeradius configuration
After batch execution, look at the database content you will see the difference between attributes in the radcheck table

droecoleman, 2019-04-18:
i would agree with @reigelgallarde they should just match since, they are just a repeat of the manual add users process. Hmmmpp.. i wonder, what could be the problem.

pihomeserver, 2019-04-18:
First check if my feeling is correct or not. If i'm correct, then you can ask about it in the Daloradius repo why the behavior is different from the manual process compared to the batch process

droecoleman, 2019-04-19:
@pihomeserver it really is, batch users and new users do not have the same attributes. i finally get this now,   Thanks man.

closed 2019-04-19 by droecoleman