pihomeserver/Kupiki-Hotspot-Script #94: Wifi with new Raspberry Pi 3B+

jeebee2013, 2018-03-20:
hi all
thanks again for providing us this fantastic script
i've just receive today the new Raspberry pi 3B+  :-)
this new model comes with new lan and wifi 
so i guess that wifi driver is no longer : nl80211 because the install script failed when it try to up Wlan0
if someone here know where to find wifi drivers name for this new raspberry, we could then update the script

il will try to found on my side and keep you informed here.

thanks.

pihomeserver, 2018-03-20:
Can you please share the last 5-10 lines of the log from the installation script to know which step is wrong ?
Maybe you don't need all controls anymore, i don't have the 3B+ yet

jeebee2013, 2018-03-20:
here is the last 30 lines of 

::: Updating /etc/hosts
::: Removing any previous sources of CoovaChilli project
::: Cloning CoovaChilli project
Cloning into 'coova-chilli'...
::: Cloning Pi Hotspot portal project
Cloning into 'portal'...
::: Starting MySql service
::: Update interface configuration (eth0)
::: Update interface configuration (wlan0)
::: Activating the WAN interface
Internet Systems Consortium DHCP Client 4.3.5
Copyright 2004-2016 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/b8:27:eb:4e:53:6f
Sending on   LPF/eth0/b8:27:eb:4e:53:6f
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 4
DHCPREQUEST of 192.168.1.102 on eth0 to 255.255.255.255 port 67
DHCPOFFER of 192.168.1.102 from 192.168.1.254
DHCPACK of 192.168.1.102 from 192.168.1.254
RTNETLINK answers: File exists
bound to 192.168.1.102 -- renewal in 19648 seconds.
::: Activating the LAN interface
RTNETLINK answers: Operation not possible due to RF-kill
ifup: failed to bring up wlan0
:::
::: Something went wrong with the last command. Please check the log file
:::

 nothing really significant in /var/log/messages..

thanks

pihomeserver, 2018-03-20:
Not related to driver i guess but to wifi configuration
Can you please provide 
- that you use Raspbian Lite (not the standard one) 
- that you use a new SD card with fresh install
- the content of /etc/network/interfaces
- result of ifconfig -a (as root)
- result of rfkill list (as root)

jeebee2013, 2018-03-20:
rasbian lite but March 2018 version (mandatory for rasp 3B+ ,  no boot with older version)
and of course fresh install on new SD
content of /etc/network/interface
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp

auto wlan0
allow-hotplug wlan0
iface wlan0 inet static
    address 192.168.10.1
    netmask 255.255.255.0
    network 192.168.10.0
    post-up echo 1 > /proc/sys/net/ipv4/ip_forward

ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.102  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::7c77:cf1:fabc:f6d5  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:4e:53:6f  txqueuelen 1000  (Ethernet)
        RX packets 160462  bytes 206769149 (197.1 MiB)
        RX errors 0  dropped 5  overruns 0  frame 0
        TX packets 94774  bytes 7132380 (6.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 570  bytes 122926 (120.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 570  bytes 122926 (120.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  broadcast 192.168.10.255
        ether b8:27:eb:1b:06:3a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

#sudo rfkill list
0: phy0: Wireless LAN
	Soft blocked: yes
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

seems that the guy's on piDiet Gituhb forum are experiencing some issue with rasp 3B+ 
https://github.com/Fourdee/DietPi/issues/1627

pihomeserver, 2018-03-20:
Ok what is strange is that the wlan0 interface is up and with the correct ip
Can you try to stop the interface and start it again with ifup and just after do an echo $? to display the return code. With the log maybe it's différent than 0 ...

jeebee2013, 2018-03-21:
hi 
Maybe a good start of investigation in new rasbian release note
2018-03-13:
* Raspberry Pi 3 B+ support
*** WiFi is disabled until wireless regulatory domain is set (Pi 3 B+ only)**
- The domain can be done through 'Raspberry Pi Configuration' (rc_gui),
'raspi-config' or by setting 'country=' to an appropriate ISO 3166
alpha2 country code in /etc/wpa_supplicant/wpa_supplicant.conf.
* Default wireless regulatory domain is now unset

i will test and keep you inform

jeebee2013, 2018-03-23:
Hi
If you set the Wifi country Code first
Everithing will work as expected
so we canc close this Issue
Please add a notice of this for Raspberry pi 3B+ users

pihomeserver, 2018-03-25:
I will set an option in the script then it will work for all
You confirm that you add it before the script was executed (and only that) ?

jeebee2013, 2018-03-26:
hello
yes , you need to set wifi country before starting the script
otherwise, wifi interface can't be up, and script will failed
hope it help.
thx

pihomeserver, 2018-03-26:
Good. I added an option with FR as default and a test just before getting up the interface
Can you please give a try to the branch 1.8.4 ?

pihomeserver, 2018-03-30:
Branch merged with Raspberry Pi 3B+ support

closed 2018-03-30 by pihomeserver