pihomeserver/Kupiki-Hotspot-Script #97: Pi won't boot after install

Grimmy347, 2018-04-04:
Ok, so I did the install like the guide said to but it got to this point:
`-----
::: Activating portal website
::: Installing the portal in Nginx folder
::: Updating Captive Portal file
::: Checking Nginx configuration file
::: Adding Freeradius in systemd startup
Synchronizing state of freeradius.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable freeradius
::: Correct configuration for Collectd daemon
::: Creating fail2ban local configuration
::: Configuring fail2ban jail rules
::: Reloading fail2ban local configuration
::: Create banner on login
::: Append script version to the banner
::: Changing banner rights
::: Activating the banner for SSH
:::
::: Starting freeradius service
::: Restarting Nginx
::: Restarting hostapd
::: Starting CoovaChilli service
::: Starting fprobe service
`

and then it froze and will not connect to the lan or wireless internet upon reboot.  Any ideas?

Grimmy347, 2018-04-04:
Actually, it turns out that it is broadcasting its own wifi as an AP but I can no longer access it through the ethernet connection and the wifi connection won't let me access anything.

pihomeserver, 2018-04-04:
What os do you use ? What if you start fprobe manually ? Did you changed some parameters ?

Grimmy347, 2018-04-05:
I changed the hotspot name, the hotspot ip and hotspot network and the country code.  I am running raspearian stretch and I have not been able to start fprobe manually because I can't log into the pi after installing.  I am working on restoring my sd card to just prior to install right now so I can try again with some different settings though.

Grimmy347, 2018-04-05:
should the hotspot network be the one on my router that the ethernet is plugged into or can it be anuthing?

Grimmy347, 2018-04-05:
ok, so if my router ip is 192.168.1.1 then in the config, hotspo_ip can be 192.168.10.1 and hotspot_network would be my router ip of 192.168.1.1?  I am not sure if I am interpreting that correctly or not.  thanks again!

pihomeserver, 2018-04-06:
Router 192.168.1.1
Raspberry Ethernet 192.168.1.X
Raspberry Wifi 192.168.10.1 (HOTSPOT_IP from the script)
Hotspot users 192.168.10.Y(HOTSPOT_NETWORK from the script)

pihomeserver, 2018-05-02:
Script updated to add a control on these parameters

closed 2018-05-02 by pihomeserver