pihomeserver/Kupiki-Hotspot-Script #98: Auto Generated User Accounts

versavius, 2018-04-08:
For environments such as restaurants, cafe's or environments where guests are likely to come and use the account for the day it would be useful if there was a cron job (or the equivalent) to auto run the following:

1. At a preset time per (Day/Week/Month) delete or disable all current user accounts
2. Generate a set of new user accounts (choose a number between 1 and 1000) in a format of User1, User2, User3 etc, with a randomised password
3. email this list to a preset email address

This allows the shop owner or receptionist to give out individual user accounts per day and old accounts are auto-deleted/disabled. It just eliminates the need to login and do this manually.

pihomeserver, 2018-04-08:
Hello
I guess it will not help but that could be made via a script by creating records in the radius database.
1) no need to disable if you set an attribute to your user that will make it valid only one day. Look at http://networkradius.com/doc/3.0.10/raddb/mods-available/expiration.html
2) create a record in `userinfo` and then another one with the password in `radcheck`
3) send your list by the way you want

Step 2 is how i create a user with the web interface i want to use in place of daloradius.

pihomeserver, 2018-04-08:
Another point in min is to have a web interface with a button 'create random user' with a max date validity