pihomeserver/Kupiki-Hotspot-Script #99: Question Pi-Hole and Kupiki ?

versavius, 2018-04-08:
Just wondered if anyone has used Pi-Hole and Pukiki on the same Pi ?

Pi-Hole is a content blocker so for a public wifi i am going to mess around with Pi-Hole, iptables and Kupiki, the Pi-Hole is more to reduce bandwith waste and to be able to block sites if certain people really take the mick.

But just curious if anyone has used this combination or some other combination for content filtering ?

pihomeserver, 2018-04-08:
I had in mind to implement Squid but i will look at Pi Hole and if it can be combined with coova

versavius, 2018-04-08:
Pi-Hole is very powerful, if you have ever used an adblocker you basically load the lists to block i.e. easylist and it blocks those domains/ip's, auto updates from those lists, so great to filter out stuff, so for a public hotspot you could filter porn/gambling/nasty websites etc.

Pi-Hole has its own admin interface so if it can work on the same Pi as Kupiki then it might not be necessary to combine it. Works great. Very simple to install on a Pi also.

pihomeserver, 2018-04-09:
Tested and will not work out of the box. I had to change the DHCP configuration but can get the portal displayed. Instead i have the Pi-hole logo.
I guess that it has to be investigated more