pihomeserver/Kupiki-Hotspot-Script #100: ERROR: could not open moddep file '/lib/modules/4.9.80-v7+/modules.dep.bin'

medmax78, 2018-04-12:
### Actual behavior

{Modprobe module (no check - useless if already loaded)
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep () could not open moddep file '/lib/modules/4.9.80-v7+/modules.dep.bin'
modprobe: FATAL: Module tun not found in directory /lib/modules/4.9.80-v7+}

### Expected behavior

I hope it will work

#### Configuration parameters (if changed in the script)

{ Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="raspberry"
 Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="Hotspot"
 Password used for the generation of the certificate
CERT_PASSWORD="raspberry"
}

#### Logs

```
:::
::: You are root - OK
:::
::: Verifying free disk space (1024 Mb) - OK
::: :::
::: ::: Configure sysctl kernel parameters
::: :::
::: Preparing log file
::: Log file already exists. Creating a backup.
:::
::: Creating the log file
:::
::: Log file created : /var/log/pihotspot.log
::: Use command 'tail -f /var/log/pihotspot.log' in a new console to get installation details
::: Updating APT config to force IPv4
::: Configuring localepurge
:::
::: Checking apt-get for upgraded packages.... done!
:::
::: There are 18 updates available for your system!
:::
::: Upgrading the packages. Please be patient.
::: Disable integrated Bluetooth support (After next reboot)
::: Remove old configuration of Coova Chilli
::: Remove old configuration of haserl
::: Remove old configuration of hostapd
::: Checking for tun module
::: Insert tun module if existing (for Raspbian Jessie Lite)
::: Modprobe module (no check - useless if already loaded)
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.9.80-v7+/modules.dep.bin'
modprobe: FATAL: Module tun not found in directory /lib/modules/4.9.80-v7+
::: Checking for tun module
::: Unable to get tun module up. Please solve before running the script again.
::: If your distribution has been upgraded you should try to reboot first.

```

### Your environment

- Hardware : Raspberry pi 3
- Operating system: {2018-03-13-raspbian-stretch-lite}

### Additional comments

Good afternoon . I have a problem in script.I tried rebooting. With the new OS version the script does not work? In folder / lib / modules /  have only  4.14.30+ and 4.14.30-v7 + 
What should I do ? In Linux, I'm not very good. Sorry for my english (Google helps)

medmax78, 2018-04-12:
After reboot 
::: Adding Avahi dependencies
::: Updating the system hostname to Hotspot
::: Updating /etc/hosts
::: Removing any previous sources of CoovaChilli project
::: Cloning CoovaChilli project
Cloning into 'coova-chilli'...
::: Cloning daloradius project
Cloning into 'daloradius'...
::: Cloning Pi Hotspot portal project
Cloning into 'portal'...
::: Starting MySql service
Failed to restart mariadb.service: Unit mariadb.service not found.
:::
::: Something went wrong with the last command. Please check the log file
:::

pihomeserver, 2018-04-12:
Can you please run `apt-get update` as root and run the script again ? It seems there is an error when trying to install the packages

medmax78, 2018-04-12:
restarted root @ raspberrypi: ~ / Kupiki-Hotspot-Script # sudo chmod + x pihotspot.sh && sudo ./pihotspot.sh
::: Starting MySql service
Failed to restart mariadb.service: Unit mariadb.service not found.
:::
::: Something went wrong with the last command. Please check the log file

medmax78, 2018-04-12:
run apt-get update and sudo chmod + x pihotspot.sh && sudo ./pihotspot.sh 

in process ....

pihomeserver, 2018-04-12:
In /etc/init.d do you have any file called with mysql or mariadb ?
I'm pretty sure that packages are not installed

medmax78, 2018-04-12:
Everything is fine ! It worked! Thank you so much ! It was necessary for each error to reboot and run apt-get update