pihomeserver/Kupiki-Hotspot-Script #101: No Login Page

R3apeR25, 2018-04-13:
### Actual behavior
No redirection over wifi and no login page on the given IP after installation.
### Expected behavior
Redirect to Login Page.
#### Configuration parameters (if changed in the script)
HOTSPOT_IP 
HOTSPOT_NETWORK 
WIFI_COUNTRY_CODE = "IN"
HOTSPOT_HTTPS = "Y"
MAC_AUTHENTICATION_ENABLED = "Y"
#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : RaspberryPI 3
- Operating system: Raspbian Strech 2018-03

### Additional comment

R3apeR25, 2018-04-13:
I did install on Strecth lite before but the same thing occurs and no there was no error executing the script, hotspot was created and dalaradius pafe is accessible but no login page when I connect the hotspot, no redirect page.

pihomeserver, 2018-04-13:
Did you create the user in daloradius with the mac address ? 
Also you asked for HTTPS hotspot. Are you sure that you device accept self signed certificates ?

pihomeserver, 2018-04-13:
Stop coova and freeradius services. Start them manually from the command line with `-f -d` option for coova and `-X` for freeradius
You will have a lot of output that could help you to see if your device is reaching coova and the freeradius or not.
Can you please check if nginx is running ? If yes, perform a curl on the ip of the hotspot from the raspberry pi. Then you should have the index.html of the portal

R3apeR25, 2018-04-13:
Ok let try.I will post asap

pihomeserver, 2018-04-13:
Also if the certificate is not mandatory remove the option. Could be the problem (or part of it)

R3apeR25, 2018-04-13:
what should be the HOTSPOT IP and NETWORK IP in my case?
```
pi@raspberrypi:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.5  netmask 255.255.255.0  broadcast 192.168.1.255
        ether b8:27:eb:cc:da:a0  txqueuelen 1000  (Ethernet)
        RX packets 18363  bytes 3933130 (3.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3307  bytes 501751 (489.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
```
pi@raspberrypi:~ $ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    0      0        0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
```

pihomeserver, 2018-04-13:
Look at issue #103

R3apeR25, 2018-04-13:
in his case IP is diiferent from mine, how could we use same ip in both cases?

so if i leave the ip to default, will it work?

R3apeR25, 2018-04-13:
previously i was using inet address as my hotspot ip and deafult ip as my network ip. Is that is the reason i am not getting login page.
```
# IP of the hotspot
HOTSPOT_IP="192.168.1.5"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="IN"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.1.0"
```

pihomeserver, 2018-04-13:
No both networks must be different. The system will then bridge both NIC

closed 2018-04-16 by R3apeR25