pihomeserver/Kupiki-Hotspot-Script #103: Network configuration

ss0028, 2018-04-13:
### Actual behavior
what should we replace for:
HOTSPOT_IP="?"
HOTSPOT_NETWORK="?"
There was no login page with our configuration.
###LOGS
[pihotspot.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/1906937/pihotspot.log)

[image]
[image]

please help me asap.

pihomeserver, 2018-04-13:
IP of the exposed Hotspot and IP of the network that will be used for the devices on the hotspot
Must be different than the network used on the eth0 interface

In you case try
HOTSPOT_IP="192.168.10.1"
HOTSPOT_NETWORK="192.168.10.0"

R3apeR25, 2018-04-13:
okay thank you very much for your time.

but, one last problem . Can't log in, i've added user from dalaradius and verified from database, but getting this error while logging in.

[image]

pihomeserver, 2018-04-13:
try to ping the 192.168.10.1 from your raspberry. Also check that coova service is up and running

ss0028, 2018-04-14:
Thankyou, now it works.

closed 2018-04-14 by pihomeserver