pihomeserver/Kupiki-Hotspot-Script #104: No Hotspot Visible at all

MuVo2k, 2018-04-14:
### Actual behavior
First of all i wanna thank you for the great Work you did
I have to say im a total Noob with Networks, Ive already tried set up Hotspot with a Captive Portal with several Tutorials( hostapd, dhcpcd ) . 
After i didnt worked out as I exspected I tried installed Kupiki-Hotspot-Script.

After the Download the Script installation told me 

Congratulation ! You now have your hotspot ready !
::: 
::: - Wifi Hotspot available : pihotspot
::: - For the user management, please connect to http://192.168.0.105/ or http://pihotspot.local/
:::   (login : administrator / password : radius)

daloradius is working

### Expected behavior
... but the Hotspot didnt show  up ,

#### Configuration parameters (if changed in the script)
WIFI_COUNTRY_CODE="ES"

LAN_WIFI_DRIVER="brcmfmac"

#### Logs

if somebody can tell copy text or the hole log file here  

### Your environment

- Hardware : Raspberry Pi3 Model B
- Operating system: Raspbian Stretch Lite 
- ssh with ubuntu mate and guake terminal

### Additional comments

I would be thankful for any advise how to setup the hotspot, or further information

pihomeserver, 2018-04-14:
it's due to the change of the driver. You will have to configure yourself hostapd if you don't use the integrated wifi chipset. When you will see it up then reboot and all should be good

MuVo2k, 2018-04-14:
It worked  ! Thanks

closed 2018-04-14 by MuVo2k