pihomeserver/Kupiki-Hotspot-Script #105: HELP - How do you imagine an update process ?

pihomeserver, 2018-04-15:
All is in the title but i would like a feedback about how you imagine an update process to avoid "old" users to reinstall every thing
The only one i have in mind is to continue to update the script and create another one that will apply only new changes

What do you think ?

reigelgallarde, 2018-04-20:
I think we can do `$ ./pihotspot.sh update`

And with this, don't start new installation but just do `apt-get update` and run some update function defined inside pihotspot.sh...

pihomeserver, 2018-04-20:
Thanks Reigel
I have something like that in mind
For each new change the main script will be updated and i will create a patch script. When you run tje update you suggest then the main script will compare release version vs installed version and apply patches one by one until both get aligned
Just the testing process will take more time and sounds also less efficient

pihomeserver, 2018-06-01:
Implemented since version 2.0.0

closed 2018-06-01 by pihomeserver