pihomeserver/Kupiki-Hotspot-Script #106: Can we do mac authentication along with username and password both?

darkprince1337, 2018-04-17:
### Actual behavior

In daloRadius management, in new user option we are getting options to register user with either username and password or mac authentication

Is there any way by which can do both. And also some machenism by which only registered users with mac address are allowed to connect to network? 

Can you help me with that?

- Hardware : RaspberryPi 3
- Operating system: Raspbian Strech

pihomeserver, 2018-04-17:
Hello
The best way is to test that. Did you activated the MAC authentication in the script before the installation ? If yes then if the MAC auth fails then you should be redirected to the portal where you can input username/password in a classic way

pihomeserver, 2018-04-17:
Look at script source to activate it after installation. It's only few lines of code.

About the lockout, i guess that's something like [this](https://www.paranoids.at/freeradius-anti-bruteforce-lockout-via-mysql/). If you want the feature, please open a new dedicated issue.

darkprince1337, 2018-04-17:
okay i'm opening new issue for non-registered MAC address blocking feature.

closed 2018-06-06 by pihomeserver