pihomeserver/Kupiki-Hotspot-Script #110: ERROR BY WIFI & CABLE

shikyo, 2018-04-29:
// CONECTED BY WIFI ON THE SERVER 
iMac-de-Jean-philippe:~ jp29640$ ping 192.168.1.115
PING 192.168.1.115 (192.168.1.115): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3

// CONECTED BY CABLE 
iMac-de-Jean-philippe:~ jp29640$ ping 192.168.1.115
PING 192.168.1.115 (192.168.1.115): 56 data bytes
64 bytes from 192.168.1.115: icmp_seq=0 ttl=64 time=1.015 ms
64 bytes from 192.168.1.115: icmp_seq=1 ttl=64 time=0.609 ms
64 bytes from 192.168.1.115: icmp_seq=2 ttl=64 time=0.610 ms

// ERROR BY CABLE WHEN ACCESS TO BROWSER TO 192.168.1.115
The last contact with the Controller (192.168.1.115:3990) failed
Error : JSON request timed out (or reply is not valid)

// CONFIGURATION
#!/bin/bash

# PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION

# Name of the log file
LOGNAME="pihotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="CLUB CROISSANT WIFI"
# IP of the hotspot
HOTSPOT_IP="192.168.1.115"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.1.115"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
# WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="nl80211"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="Y"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="Y"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
# Set value to Y or N
NETFLOW_ENABLED="Y"
# Define how long Netflow logs will be stored
# Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
# Accpeted time scales are w (weeks) d (days) H (hours).
# A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
# By default data are stored 365 days (value set to 365d)
NETFLOW_LOGS_DAYS="365d"
# Enable/Disable MAC authentication
# Set value to Y or N
MAC_AUTHENTICATION_ENABLED="N"
# Password for MAC authentication. Should be changed.
MAC_AUTHENTICATION_PASSWORD="123456"
# Install web frontend of Kupiki Hotspot
# Set value to Y or N
INSTALL_KUPIKI_ADMIN=Y

pihomeserver, 2018-04-30:
Hotspot network is wrong. You set it as an ip. Set it it to 192.168.1.0That also means that the network connected to the Ethernet is different
-------- Message d'origine --------De : shikyo <notifications@github.com> Date : 30/04/2018  01:53  (GMT+01:00) À : pihomeserver/Kupiki-Hotspot-Script <Kupiki-Hotspot-Script@noreply.github.com> Cc : Subscribed <subscribed@noreply.github.com> Objet : [pihomeserver/Kupiki-Hotspot-Script] ERROR BY WIFI & CABLE (#110) 
// CONECTED BY WIFI ON THE SERVER

iMac-de-Jean-philippe:~ jp29640$ ping 192.168.1.115

PING 192.168.1.115 (192.168.1.115): 56 data bytes

Request timeout for icmp_seq 0

Request timeout for icmp_seq 1

Request timeout for icmp_seq 2

Request timeout for icmp_seq 3
// CONECTED BY CABLE

iMac-de-Jean-philippe:~ jp29640$ ping 192.168.1.115

PING 192.168.1.115 (192.168.1.115): 56 data bytes

64 bytes from 192.168.1.115: icmp_seq=0 ttl=64 time=1.015 ms

64 bytes from 192.168.1.115: icmp_seq=1 ttl=64 time=0.609 ms

64 bytes from 192.168.1.115: icmp_seq=2 ttl=64 time=0.610 ms
// ERROR BY CABLE WHEN ACCESS TO BROWSER TO 192.168.1.115

The last contact with the Controller (192.168.1.115:3990) failed

Error : JSON request timed out (or reply is not valid)
// CONFIGURATION

#!/bin/bash
PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION
Name of the log file
LOGNAME="pihotspot.log"
Path where the logfile will be stored
be sure to add a / at the end of the path
LOGPATH="/var/log/"
Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="CLUB CROISSANT WIFI"
IP of the hotspot
HOTSPOT_IP="192.168.1.115"
Wi-fi code country. Use above link to find yours
https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
Use HTTPS to connect to web portal
Set value to Y or N
HOTSPOT_HTTPS="N"
Network where the hotspot is located
HOTSPOT_NETWORK="192.168.1.115"
Secret word for FreeRadius
FREERADIUS_SECRETKEY=cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1
WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'
LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
Wifi driver
LAN_WIFI_DRIVER="nl80211"
Install Haserl (required if you want to use the default Coova Portal)
Set value to Y or N
HASERL_INSTALL="Y"
Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
Make Avahi optional
Set value to Y or N
AVAHI_INSTALL="Y"
Install Daloradius Portal (compatible with FR2 only in theory)
Set value to Y or N
DALORADIUS_INSTALL="Y"
Enable/Disable Bluetooth
Set value to Y or N
BLUETOOTH_ENABLED="N"
Enable/Disable fail2ban to protect server from unwanted access
Set value to Y or N
FAIL2BAN_ENABLED="Y"
Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
Set value to Y or N
NETFLOW_ENABLED="Y"
Define how long Netflow logs will be stored
Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
Accpeted time scales are w (weeks) d (days) H (hours).
A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
By default data are stored 365 days (value set to 365d)
NETFLOW_LOGS_DAYS="365d"
Enable/Disable MAC authentication
Set value to Y or N
MAC_AUTHENTICATION_ENABLED="N"
Password for MAC authentication. Should be changed.
MAC_AUTHENTICATION_PASSWORD="123456"
Install web frontend of Kupiki Hotspot
Set value to Y or N
INSTALL_KUPIKI_ADMIN=Y

shikyo, 2018-04-30:
I set like you said me and is already not working. 

Impossible to access to 192.168.1.0 or 192.168.1.115 by the wifi from the hotspot. I see you're French (I'm also French) if we could speak in French it would be more easier.

#!/bin/bash

# PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION

# Name of the log file
LOGNAME="pihotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="CLUB-CROISSANT-WIFI"
# IP of the hotspot
HOTSPOT_IP="192.168.1.0"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.1.115"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
# WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="nl80211"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="Y"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="Y"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
# Set value to Y or N
NETFLOW_ENABLED="Y"
# Define how long Netflow logs will be stored
# Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
# Accpeted time scales are w (weeks) d (days) H (hours).
# A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
# By default data are stored 365 days (value set to 365d)
NETFLOW_LOGS_DAYS="365d"
# Enable/Disable MAC authentication
# Set value to Y or N
MAC_AUTHENTICATION_ENABLED="N"
# Password for MAC authentication. Should be changed.
MAC_AUTHENTICATION_PASSWORD="123456"
# Install web frontend of Kupiki Hotspot
# Set value to Y or N
INSTALL_KUPIKI_ADMIN=Y

shikyo, 2018-04-30:
Faut t'il configuré aussi quelque chose sur le routeur ? et d'un autre coté il est possible de modifié les paramètres une fois le script installé ?

pihomeserver, 2018-04-30:
Rien sur le routeur ou la box du FAI

Oui mais ca devient plus compliqué. Le script ne permet pas de mise à jour

shikyo, 2018-04-30:
C'est juste suite au informations ci-dessus ?

# Name of the log file
LOGNAME="pihotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="CROISSANTWIFI"
# IP of the hotspot
HOTSPOT_IP="192.168.10.1"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.10.255"

pihomeserver, 2018-04-30:
Remplace le 255 par 0 et ca sera bon

shikyo, 2018-05-01:
Je te remercie pour ton aide sa marche du tonnerre ! Une question en plus:
Kupiki Hotspot Admin est compatible avec le hotspot ou sont-t'il tout deux différent.

D'un autre coté, comment puis t'offrir une donation ?

pihomeserver, 2018-05-01:
Bonne nouvelle
L'Admin est compatible avec le hotspot ca doit aider à administrer mais attention c'est en cours de développement donc plein de bugs 

Pour la donation il y a un lien en bas de la page Github ;-)

closed 2018-05-02 by pihomeserver