pihomeserver/Kupiki-Hotspot-Script #111: Server certificate failed

Megafab, 2018-05-15:
### Actual behavior

Bonjour,
Sur un Raspian Lite , j'ai le message suivant : 
fatal: unable to access 'https://github.com/pihomeserver/Kupiki-Hotspot-Script.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

Alors que jusque ici cela marchait parfaitement.

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : {Please write here}
- Operating system: {Please write here}

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2018-05-15:
Essayez un apt-get update puis apt-get upgrade et enfin de relancer le script

Megafab, 2018-05-15:
Bonjour,

C'est pareil.
C'est une Raspbian que j'ai réinstallé, mis a jour.La semaine dernière cela fonctionnait parfaitement.

Megafab, 2018-05-15:
Oups pardon de ma faute, il n'était pas à l'heure ...

closed 2018-05-15 by Megafab