pihomeserver/Kupiki-Hotspot-Script #112: Pas d'accès à internet

Megafab, 2018-05-15:
### Actual behavior
Bonjour,

Un autre soucis ( mais la je ne trouve pas) , la wifi et bien détecté, mais pas d'accès internet.
Je n'ai rien fait d'autre que les mises à jour, et lancer le script.

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : Raspberry 3+
- Operating system: Raspbian Lite

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2018-05-15:
Quelles sont les 20 dernières lignes de la log ? Sans plus d'info je ne peux pas comprendre le problème
Quelle est la config réseau ? En SSH est-ce qu'un ping google abouti ? Un user a bien été créé dans freeradius ?

Megafab, 2018-05-15:
La config réseau:

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.20.8.210  netmask 255.255.0.0  broadcast 10.20.255.255
        ether b8:27:eb:4e:12:ce  txqueuelen 1000  (Ethernet)
        RX packets 29195  bytes 5317497 (5.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1593  bytes 430748 (420.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 7  bytes 1269 (1.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7  bytes 1269 (1.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=81<UP,POINTOPOINT,RUNNING>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  destination 192.168.10.1
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 236  bytes 18164 (17.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 236  bytes 24772 (24.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4419<UP,BROADCAST,RUNNING,PROMISC,MULTICAST>  mtu 1500
        ether b8:27:eb:1b:47:9b  txqueuelen 1000  (Ethernet)
        RX packets 346  bytes 26154 (25.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 76  bytes 7926 (7.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

En ssh je ping bien Google ,et un utilisateur est bien créé:
[image]

Megafab, 2018-05-15:
Lorsque je veux me connecter au hotspot j'ai le message : Le réseau Wi-fi ne dispose d'aucun accès à internet

pihomeserver, 2018-05-15:
Quel est le device utilisé pour se connecter ? Ou est affiché le message ?

pihomeserver, 2018-05-15:
Le portail captif s'affiche bien ?Si oui l'authentification fonctionne avec un message de succès ?

Megafab, 2018-05-15:
Sur un téléphone, nexus.Lorsque que je choisis le hotspot
Le portail captif ne s'affiche pas.

pihomeserver, 2018-05-15:
Le portail est exposé via un serveur Nginx. Est-ce que le service tourne bien ?
```
service nginx status
```
Est-ce que l'installation s'est bien terminée au niveau de la log dans /var/log/ ?
```
tail -20 /var/log/pihotspot.log
```

Megafab, 2018-05-15:
root@watwifi:/home/pi# /etc/init.d/freeradius restart
[ ok ] Restarting freeradius (via systemctl): freeradius.service.
root@watwifi:/home/pi# service nginx status
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-05-15 16:40:37 CEST; 3min 12s ago
     Docs: man:nginx(8)
  Process: 829 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 565 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
 Main PID: 831 (nginx)
   CGroup: /system.slice/nginx.service
           ├─831 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─832 nginx: worker process
           ├─833 nginx: worker process
           ├─834 nginx: worker process
           └─835 nginx: worker process

mai 15 16:40:35 watwifi systemd[1]: Starting A high performance web server and a reverse proxy server...
mai 15 16:40:37 watwifi systemd[1]: Started A high performance web server and a reverse proxy server.

Megafab, 2018-05-15:
root@watwifi:/home/pi# tail -20 /var/log/pihotspot.log
::: Changing banner rights
::: Activating the banner for SSH
:::
::: Starting freeradius service
::: Restarting Nginx
::: Restarting hostapd
::: Starting CoovaChilli service
::: Starting fprobe service
::: Reloading units for systemctl
::: Starting nfdump service
::: Reload configuration for SSH service
::: Checking if interface tun0 has been created by CoovaChilli
tun0: flags=81<UP,POINTOPOINT,RUNNING>  mtu 1500
:::
:::
::: Congratulation ! You now have your hotspot ready !
:::
::: - Wifi Hotspot available : watwifi
::: - For the user management, please connect to http://10.20.8.210/ or http://watwifi.local/
:::   (login : administrator / password : radius)

pihomeserver, 2018-05-15:
Quel est le contenu de la config Nginx associée au portail ?
```
cat /etc/nginx/sites-enabled/portal
```

pihomeserver, 2018-05-15:
Essayez de faire un stop start de coova. La connexion du user test via le portail fonctionne ? Avec l'écran de validation ?

closed 2018-05-18 by Megafab

pihomeserver, 2018-05-18:
Quelle a été la résolution ?