pihomeserver/Kupiki-Hotspot-Script #113: Adding Hotspot

Megafab, 2018-05-18:
### Actual behavior

Hi

my hotspot works perfectly.
now i'm trying to add a new hotspot via daloradius, and i'm blocking it a bit.

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : Raspberry 3+
- Operating system: Raspbian

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2018-05-18:
I guess that you want only one instance of Freeradius to administrate your users ?
If so it's not possible via the script yet. You have to change manually the IP address in the Coova configuration and restart the service
Also for optimization purpose you can disable the unused Freeradius instance

Megafab, 2018-05-18:
ok, dans le fichier default de chilli ? A quelle ligne ?

pihomeserver, 2018-05-18:
In the HS_RADIUS parameter in the file called /etc/chilli/defaults

Never tested

Megafab, 2018-05-18:
The goal being to add more hotspot, and roaming

pihomeserver, 2018-05-18:
That's the purpose of the parameter. Now if you analyze the structure you will see that it implies that your radius server must be reachable from Internet and/or Intranet. So with the security level according to the risk that you will have to expose the authentication server

Megafab, 2018-05-18:
[image]
I added the ip of my hotspot in the file.

Megafab, 2018-05-18:
and add the ip of my hotspot to the clients.conf file
[image]
but no issue

pihomeserver, 2018-05-18:
Not sure : 
- Hotspot A is your main hotspot with
   - Coova
   - Freeradius
- Hotspot B is your additional hotspot with
   - Coova
   - Freeradius

On the hotspot B, disable freeradius and change in coova configuration the HS_RADIUS IP

Megafab, 2018-05-18:
ok I test all this

Megafab, 2018-05-18:
ok it works.
but would it be possible to connect to a single portal?

pihomeserver, 2018-05-18:
Single portal ?

Megafab, 2018-05-18:
the authentication is done well on a single freeradius, but I must connect on each hotspot

pihomeserver, 2018-05-18:
Sure, you connect on a signal. Now if you want to share the portal exposed on your Hotspot A even if you use the Hotspot B signal then you can try to have a reverse proxy on the B that will push all HTTP request on the B portal to the A portal.
Nginx is already installed so you can try to change the configuration portal in /etc/nginx/sites-available/portal with proxy_pass.
Again never tried that configuration of the hotspot

pihomeserver, 2018-05-18:
Also if you want to extend the signal for a huge area or floors then may be another way to do so is to use Wifi repeater. Depends of your budget/objectives/etc.

vanrion, 2018-08-20:
Hi, is it possible to add another eth3 for another AP? 
For ex: eth1 is for WAN
            eth2 is for AP1
            eth3 is for AP2 to cover the dead spot

pihomeserver, 2018-08-20:
@vanrion new issue #126 opened