pihomeserver/Kupiki-Hotspot-Script #115: Connect Daloradius throught hotspot

CBeyssier, 2018-05-26:
### Actual behavior

{Please write here}

### Expected behavior

Good morning.
Please is it possible to connect to Daloradius throught hotspot if Raspberry don't have display ?

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : {Please write here}
- Operating system: {Please write here}

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2018-05-26:
Daloradius is exposed as a website so you don't need display
Also for security reasons you can not access daloradius from the hotspot. Only from the lan interface

CBeyssier, 2018-05-29:
Ok thank you.
I tried to install with this script on Raspberry PI 3B+ with Raspbian.
Because Minibian not available for this Raspberry yet.
Some problem de solves.
I'm discovering shell and script, I will take time to understand all but this is my challenge.
I will come back here to give more info if I'll have.

Because I start few times the script, and look into log to see problems, what I can say first it's I modify one instruction where one folder is created, to avoid error if this folder already exist (bescause of starting few time the same script). If already created, error don't come now.

Step by step ...
Excuse me but I'm really new in all this.
Regards.

Christophe

pihomeserver, 2018-05-29:
Please share the update you made to improve the script. Also be sure to use Raspbian Lite and not the full version with GUI

CBeyssier, 2018-05-29:
Ok will try with raspbian lite.

Improvement (if you agree):
Line 489 mkdir -p /var/local/kupiki
to avoid error if Kupiki admin folder already exist.

Thanks.
Christophe