pihomeserver/Kupiki-Hotspot-Script #116: Problem with pihotspot connection page on tablet

CBeyssier, 2018-05-29:
### Actual behavior

Al installed, wifi hotspot found with tablet
Kupiki ask user login and password (created in daloRADIUS with laptop on LAN)
Fillin this
Accept term and licence and Connect
But after few seconds:
"The last contact with the Controller (192.168.19.1:3990) failed
Error: JSON request timed out (or replys is invalid)

### Expected behavior

{Please write here}

#### Configuration parameters not changed in script

#### Logs

```

```

### Your environment

- Hardware : Raspberry PI3B+
- Operating system: Raspbian lite (04/2018)

### Additional comments

pihomeserver, 2018-05-29:
What is the IP of your eth0 interface and the IP of your wlan interface ? Seems that the IP configuration is wrong because you changed it in the script. Please provide updated parameters

pihomeserver, 2018-05-31:
I am testing the next upgrade of the script and get the same issue. Did you solve it ?

closed 2018-06-01 by pihomeserver

pihomeserver, 2018-06-02:
Yes if you reboot after or do a stop start of freeradius service