pihomeserver/Kupiki-Hotspot-Script #117: JSON problem ... again ... sorry

CBeyssier, 2018-06-02:
Good afternoon again,

I decide to re-install the script, after your update.
Same problem with JSON

The last contact with the controler (192.168.10.1: 3990) failed
Error: JSON request timed out (or reply is not valid).

When I reboot I saw dhcpcd failed to start, but perhaps it's normal I don't know.

Perhaps this is because my internet connection is not stable, I at sea on merchant vessel. Loool.
From a laptot I can configure users etc ...
From tablet I succed to found wifi, I reach the page with username and password, but never possible to connect.
On terminal (raspberry) with host www.google.fr I obtain ip adresse but not every time I try. (because of my connection I think.
What I plan to do, when I'll be alongside in an harbor with better inernet, I'll try again, and perhaps re-install all with a brand new image of Raspbian stretch (the first image I start with).

I'll stop for the moment, thank you.
Very interesting to "enter" or try to "enter" in a Raspberry ...
Christophe

Raspberry 3B+
SD Card 8 Go
Raspbian strech
and ... me (new guy in all this)

CBeyssier, 2018-06-02:
For info, I didn't change anything in the script.

pihomeserver, 2018-06-02:
As explained in the #87 the best way to debug is to stop freeradius and chilli services and start them in debug mode to see what happens when you try to connect. 
I tested yesterday the script from scratch and got no issue so i guess that there is an issue somewhere else ... and we will find it ;-)

pihomeserver, 2018-06-06:
COA updated in the release 2.0.2
Please use the updater if you have installed release greater than 2.0.0

closed 2018-06-06 by pihomeserver