pihomeserver/Kupiki-Hotspot-Script #119: SSH from within Captive Portal

pihomeserver, 2018-07-09:
### Actual behavior

SSH Access from captive portal to Pi is blocked

### Expected behavior

 I used to be able to ssh into the pi from within the captive portal with v1.4.3
But with the latest version I am not able to ssh into the pi any more from within the captive portal. 
This is a great help because I can’t have physical access the eth0 interface.

pihomeserver, 2018-07-09:
Now there is a rule that blocks any access to local network from the captive portal for security reasons. The rules are defined at lines 734-735 of the script :
```
ipt -I FORWARD 1 -i $TUNTAP -d $LOCAL_LAN -j DROP
ipt -I INPUT 1 -i $TUNTAP -d $LOCAL_IP -j DROP
```
If you know what you are doing then you can remove them, stop/start coova and you should ssh again.

vangelishat, 2018-07-09:
Hi, the rules keep returning after stop/start coova even though I have removed the lines mentioned above manually and from chilli/up.sh
Still struggling to ssh, below is my iptables list
```
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-N f2b-sshd
-A INPUT -p tcp -m multiport --dports 22 -j f2b-sshd
-A INPUT -i wlan0 -j DROP
-A INPUT -d 192.168.10.1/32 -i tun0 -p icmp -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p udp -m udp --dport 53 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 255.255.255.255/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 80 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 4990 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 3990 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -j DROP
-A FORWARD -i tun0 -o eth0 -j ACCEPT
-A FORWARD -i tun0 ! -o eth0 -j DROP
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o tun0 -j ACCEPT
-A FORWARD -i tun0 -j ACCEPT
-A FORWARD -o wlan0 -j DROP
-A FORWARD -i wlan0 -j DROP
-A f2b-sshd -j RETURN
```

pihomeserver, 2018-07-09:
Please share last 20 lines of up.sh

vangelishat, 2018-07-10:
Hi, here they are:

```
            # Help out MTU issues with PPPoE or Mesh
            ipt -I FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
            ipt -I FORWARD -t mangle -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
            
            [ "$HS_LAN_ACCESS" != "on" -a "$HS_LAN_ACCESS" != "allow" ] && \
                ipt -I FORWARD -i $TUNTAP \! -o $HS_WANIF -j DROP
            
            ipt -I FORWARD -i $TUNTAP -o $HS_WANIF -j ACCEPT
            
            [ "$HS_LOCAL_DNS" = "on" ] && \
                ipt -I PREROUTING -t nat -i $TUNTAP -p udp --dport 53 -j DNAT --to-destination $ADDR
        fi
    fi

    # site specific stuff optional
    [ -e /etc/chilli/ipup.sh ] && . /etc/chilli/ipup.sh
}

FLOCK=$(which flock)
if [ -n "$FLOCK" ] && [ -z "$LOCKED_FILE" ]
then
    export LOCKED_FILE=/tmp/.chilli-flock
    flock -x $LOCKED_FILE -c "$0 $@"
else
    run_up
fi
ipt -I POSTROUTING -t nat -o $HS_WANIF -j MASQUERADE
```

pihomeserver, 2018-07-10:
Set HS_LAN_ACCESS to off in the coova default configuration. Stop and start the service and test again
I should use that instead a forces rule definition ...

vangelishat, 2018-07-10:
Ok so, I managed to get rid of `FORWARD -i tun0 ! -o eth0 -j DROP` but by setting HS_LAN_ACCESS to **_on_** not off.
I can't get rid of `INPUT -d 192.168.10.1/32 -i tun0 -j DROP`.
After deleting it from the iptables and saving and then stop/start coova it reappears in the iptables.

pihomeserver, 2018-07-11:
Line 35 you can find : `ipt_in --dst $ADDR -j DROP`
ipt_in creates the input rule. Put in comment that line and redo the stop/start

vangelishat, 2018-07-11:
ok that got rid of `INPUT -d 192.168.10.1/32 -i tun0 -j DROP` but I still can't ssh in through wifi captive portal.
Here is my iptables list:
```
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-N f2b-sshd
-A INPUT -p tcp -m multiport --dports 22 -j f2b-sshd
-A INPUT -i wlan0 -j DROP
-A INPUT -d 192.168.10.1/32 -i tun0 -p icmp -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p udp -m udp --dport 53 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 255.255.255.255/32 -i tun0 -p udp -m udp --dport 67:68 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 80 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 4990 -j ACCEPT
-A INPUT -d 192.168.10.1/32 -i tun0 -p tcp -m tcp --dport 3990 -j ACCEPT
-A FORWARD -i tun0 -o eth0 -j ACCEPT
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -o tun0 -j ACCEPT
-A FORWARD -i tun0 -j ACCEPT
-A FORWARD -o wlan0 -j DROP
-A FORWARD -i wlan0 -j DROP
-A f2b-sshd -j RETURN
```

pihomeserver, 2018-07-11:
Can you try to stop fail2ban service ? And stop start etc

vangelishat, 2018-07-11:
I tried `ssh pi@192.168.10.1` and it worked! Seems to not work with `ssh pi@hostname.local`.
This worked without the need to stop fail2ban, but only works if you comment out line 35: `ipt_in --dst $ADDR -j DROP` in up.sh and change/add `HS_LAN_ACCESS=on` in the coova default config.

Thank you for your help!

pihomeserver, 2018-07-11:
Ok thanks. I guess that avahi is not exposing the service on the tun0 interface

Have to check in coova if there is an option to include the gateway IP in the LAN

tobeedelafuente, 2018-09-18:
greetings, I also want to SSH on the Captive portal so I could still access my RPi. Just wondering about the "coova default config" Is it "/etc/chilli/defaults"?.

@vangelishat basically the two files need to be changed in order for SSH to work correct? Thanks in advance for the help

closed 2018-11-07 by pihomeserver