pihomeserver/Kupiki-Hotspot-Script #122: Is possible to split the code?

desevensa, 2018-07-13:
I wonder if is possible to make the installation in _offline_ mode and for that I guess the best way is to have 2 scripts. The 1st in order to download the packages and files necessary for the installation, and the 2nd for the installation of those packages with PRi not plugged to Internet.

Those scripts are not my better skill but i guess the key for that could be something like:
```
**1st**
#/bin/bash
apt-get --download-only [packages]
wget [options] [files]
.
.
.

**2nd**
#/bin/bash
apt-get install [packages] #Downloaded packages
#code for compile the rest.
.
.
.
```
I would like to get a lead for accomplish this goal.

pihomeserver, 2018-07-14:
It's nearly what is already made in the existing script where it downloads all possible packages before doing the installation.
You can copy paste the script, remove installation steps in the first, remove the download steps in the second.
I guess a already installed image could be useful

desevensa, 2018-07-27:
Thanks for your answer, i will try it.

Trying options, accidentally i think i found a little bug in the **install_dependent_packages** function (lines 335-348):
If the package installation gets any error (i.e. a package not found) the function breaks with no installation and the script just continues. It gives lately an error because, in the case of mariadb, the service cannot stop because dont exist.

Could be possible catch that errors to stop the script and know what went wrong with installation.

pihomeserver, 2018-07-27:
Thanks for the issue report. Will fix it in the next release of the script

pihomeserver, 2018-11-07:
Integrated in v2.0.7 release

closed 2018-11-07 by pihomeserver