pihomeserver/Kupiki-Hotspot-Script #125: Batch users json request time out

droecoleman, 2018-07-30:
batch users json request time out. 
take note i did not change anything. 
i also refer to other previous suggestions created issues.  But it did not help

pihomeserver, 2018-07-30:
Please provide the result of ifconfig -a command to check ip allocation on all interfaces

droecoleman, 2018-07-31:
[image]

here it is.

droecoleman, 2018-07-31:
[image]

[image]

[image]

here it is.

pihomeserver, 2018-07-31:
I Don't get where the issue occurs. In Dalo or the portal ?

droecoleman, 2018-07-31:
im sorry the error occurs in the portal.

pihomeserver, 2018-07-31:
And the message that you get from the portal ?
Do you see all the users in daloradius ?

droecoleman, 2018-07-31:
yes i can see all the users  in daloradius. however when i try to log them in to the portal I always get the error json request time out

pihomeserver, 2018-07-31:
Can you try with a user created manually with a clear password attribute ? The bath uses the user password attribute and it seems to not work as expected

droecoleman, 2018-07-31:
i have done creating without attribute but still the error is there..

pihomeserver, 2018-07-31:
Does chilli process running ?
You need at least one attribute cleartext-password to define the user  password.

droecoleman, 2018-07-31:
yes it is running.. but freeradius  

[image]

droecoleman, 2018-07-31:
freeradius is now running..

then this

[image]

droecoleman, 2018-07-31:
[image]

pihomeserver, 2018-07-31:
Same here and should not be an issue
Try to create a user test with a check attribute "Cleartext-Password", operant set to ":=" and value to "test". Then you should be able to connect with username test and password test

pihomeserver, 2018-07-31:
Are you sure that you stopped freeradius before running it again ? The message says that port 1700 is already used ...

droecoleman, 2018-07-31:
yes, i stopped freeradius and that is the output

pihomeserver, 2018-07-31:
After a reboot, is freeradius running ? Only freeradius is using the port 1700 so that means that you have multiple freeradius running and of course that's not the expected behavior.
You confirm using the Raspbian Lite and not the full Raspbian OS ?

droecoleman, 2018-07-31:
i am now getting this

[image]

pihomeserver, 2018-08-01:
Keep that window open, then create the user as explained below
```
Try to create a user test with a check attribute "Cleartext-Password", operant set to ":=" and value to "test". Then you should be able to connect with username test and password test
```
And check the connectivity with the radtest command
```
radtest test test 127.0.0.1 1812 <my_secret_key>
```
The secret key can be found with the command 
```
cat /etc/freeradius/3.0/clients.conf | grep secret | grep -v '[\s]*#' | head -1 | cut -d '=' -f 2 | awk '{$1=$1};1'
```

pihomeserver, 2018-08-02:
In addition : 

> It is important to understand the difference between the  User-Password
       and Cleartext-Password attributes.  The Cleartext-Password attribute is
       the "known good" password for the user.	Simply supplying  the  Cleart-
       ext-Password  to  the server will result in most authentication methods
       working.  The User-Password attribute is the password as  typed	in  by
       the  user  on  their  private  machine.	 The two are not the same, and
       should be treated very differently.  That is, you should generally  not
       use the User-Password attribute anywhere in the RADIUS configuration.

[Source](https://freeradius.org/radiusd/man/rlm_pap.txt)

Then why Daloradius is using User-Password instead Cleartext-password, i don't know why yet.

desevensa, 2019-01-31:
@droecoleman : I realized about issue 87, if for some reason the install stops after freeradius get installed (Ctrl+c or power outage, or any other reason), Re-launch the script ends in the same error because in line 689 `sed -i "s/testing123/$FREERADIUS_SECRETKEY/g" /etc/freeradius/3.0/clients.conf` "testing123" already changed and those keys wont be the same. 
If you have to run again the script you must have this in mind to avoid that issue.

pihomeserver, 2019-02-01:
@desevensa I will change that. Thank you

droecoleman, 2019-04-17:
i did a fresh install, but then the problem now, the batch users are not recognized by the data base. if you  try using the batch add users, it will be created successful but it wont be recognize when the passwords not recognized

closed 2019-05-12 by droecoleman