pihomeserver/Kupiki-Hotspot-Script #127: How to access connection history log

Manifact, 2018-08-29:
Hi,

I'm wondering how to access the connection history log which allow to identify what user was connected with what equipment.
To be very specific, I'm talking about the things spoken in the following link : https://www.cdse.fr/wifi-et-conservation-des-donnees (in french)

Here is a translation :
"It is about :
Informations allowing to identify users (IP address, phone number, e-mail);
Informations about their "end" communications equipments;
Date, time, duration and technical features of each communication;
Extra services data asked or used and their suppliers;
Data allowing to identify recipients of the communication."

Thanks

Thibault

Manifact, 2018-08-29:
Thanks for your reply.
When I try this command as root, I have the following message : 

> "Date first seen          Duration Proto      Src IP Addr:Port          Dst IP Addr:Port   Flags Tos  Packets    Bytes      pps      bps    Bpp Flows
> No matched flows"

What do I do wrong ?

Thanks.

Manifact, 2018-08-30:
I created a new user using Daloradius,
I connected using it and browsed several pages,
I checked the /var/cache/nfdump folder and I have no folder of today. The last folder is of the 28/08/18.
The nfdump command is still returning the same thing.

Here is what return `ls -l *` : 

> total 24
> drwxr-xr-x 2 root root  4096 Aug 28 00:04 2018-08-27
> drwxr-xr-x 2 root root 12288 Aug 30 12:01 2018-08-28
> -rw-r--r-- 1 root root   276 Aug 27 18:22 nfcapd.201808271820
> -rw-r--r-- 1 root root     0 Aug 28 17:49 nfcapd.current.550
> -rw-r--r-- 1 root root   276 Aug 28 17:55 nfcapd.current.551

Thanks for the help.

pihomeserver, 2018-08-31:
Can you please run the commands : `service probe status` and `service nfdump status`

Manifact, 2018-08-31:
Sure,

Here are the results : 
`service probe status`

> Unit probe.service could not be found.

`service nfdump status`

> ● nfdump.service - netflow capture daemon
  >    Loaded: loaded (/lib/systemd/system/nfdump.service; enabled; vendor preset: enabled)
  >    Active: inactive (dead) since Fri 2018-08-31 16:06:56 UTC; 5h 38min ago
 >   Process: 555 ExecStart=/usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e (code=exited, status=0/SUCCESS)
>  Main PID: 555 (code=exited, status=0/SUCCESS)
  >    CGroup: /system.slice/nfdump.service
            >            ├─589 /usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e
            >            └─590 /usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e
> 
> Aug 31 16:06:56 kazlab nfcapd[590]: Launcher: Sleeping
> Aug 31 16:20:34 kazlab nfcapd[589]: Ident: 'none' Flows: 0, Packets: 0, Bytes: 0, Sequence Errors: 0, Bad Packets: 0
> Aug 31 16:20:34 kazlab nfcapd[589]: Signal launcher
> Aug 31 16:20:34 kazlab nfcapd[589]: Total ignored packets: 0
> Aug 31 16:20:34 kazlab nfcapd[590]: Launcher: Wakeup
> Aug 31 16:20:34 kazlab nfcapd[590]: Run expire on '/var/cache/nfdump'
> Aug 31 16:20:34 kazlab nfcapd[590]: Limits: Filesize <none>, Lifetime 31536000 = 52.1 weeks, Watermark: 95%
> Aug 31 16:20:34 kazlab nfcapd[590]: Current size: 1163264 = 1.1 MB, Current lifetime: 337616 = 3.9 days, Number of files: 284
> Aug 31 16:20:34 kazlab nfcapd[590]: expire completed - nothing to expire.
> Aug 31 16:20:34 kazlab nfcapd[590]: Launcher: Sleeping

When I try ` service nfdump start` I have : 

> ● nfdump.service - netflow capture daemon
>    Loaded: loaded (/lib/systemd/system/nfdump.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Fri 2018-08-31 21:46:01 UTC; 10s ago
>   Process: 2117 ExecStart=/usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e (code=exited, status=255)
>  Main PID: 2117 (code=exited, status=255)
>    CGroup: /system.slice/nfdump.service
>            ├─589 /usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e
>            └─590 /usr/bin/nfcapd -D -l /var/cache/nfdump -P /var/run/nfcapd.pid -p 2055 -z -S 7 -e
> 
> Aug 31 21:46:01 kazlab systemd[1]: Started netflow capture daemon.
> Aug 31 21:46:01 kazlab nfcapd[2117]: Could not open the requested socket: Address already in use
> Aug 31 21:46:01 kazlab nfcapd[2117]: Terminated due to errors.
> Aug 31 21:46:01 kazlab systemd[1]: nfdump.service: Main process exited, code=exited, status=255/n/a
> Aug 31 21:46:01 kazlab systemd[1]: nfdump.service: Unit entered failed state.
> Aug 31 21:46:01 kazlab systemd[1]: nfdump.service: Failed with result 'exit-code'.

Doesn't looks good to me ... Any idea ?

Thanks.

pihomeserver, 2018-09-02:
Sorry the service is fprobe, not probe (thanks auto completion ...)
The nfdump service seems to be ok. If you look at processes running from the command line and using ps you should find them
Do you confirm that the running fprobe process is  (with `ps -ef | grep fprobe`)
```
/usr/sbin/fprobe -itun0 -fip 127.0.0.1:2055
```

pihomeserver, 2018-09-02:
Here is my output of the command `nfdump -R /var/cache/nfdump -o extended -a `
```
Date first seen          Duration Proto      Src IP Addr:Port          Dst IP Addr:Port   Flags Tos  Packets    Bytes      pps      bps    Bpp Flows
2018-08-31 19:38:39.189     0.000 UDP       192.168.10.2:1376  ->   208.67.222.222:53    ......   0        1       60        0        0     60     1
2018-08-31 19:34:38.712     0.597 TCP      169.61.101.86:443   ->     192.168.10.2:33999 .AP.S.   0        9      896       15    12006     99     1
2018-08-31 19:41:23.952     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:5276  ......   0        1       99        0        0     99     1
2018-08-31 19:32:01.734     0.860 TCP     172.217.22.106:443   ->     192.168.10.2:44045 .A...F   0        4      208        4     1934     52     1
2018-08-31 19:41:40.182     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:9292  ......   0        1       89        0        0     89     1
2018-08-31 20:04:56.186     0.000 UDP     208.67.220.220:53    ->     192.168.10.2:55785 ......   0        1       76        0        0     76     1
2018-08-31 20:05:09.831     0.000 TCP       192.168.10.2:35964 ->      54.93.152.4:443   .AP..F   0        1       83        0        0     83     1
2018-08-31 19:41:35.904     0.258 TCP     172.217.16.206:443   ->     192.168.10.2:56974 .AP.S.   0       11     1247       42    38666    113     1
2018-08-31 19:41:35.894   190.766 TCP       192.168.10.2:56974 ->   172.217.16.206:443   .APRS.   0       11     1958        0       82    178     2
2018-08-31 19:38:45.839     0.097 TCP     173.194.76.109:993   ->     192.168.10.2:34909 .A...F   0        3      156       30    12865     52     1
2018-08-31 20:05:16.141     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:49318 ......   0        1      336        0        0    336     1
2018-08-31 19:38:33.006  1216.340 TCP       192.168.10.2:60608 ->   172.217.16.196:443   .AP.S.   0       11     1881        0       12    171     3
2018-08-31 19:39:37.016     0.000 UDP       192.168.10.2:50766 ->   208.67.220.220:53    ......   0        1       60        0        0     60     1
2018-08-31 19:34:38.607     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:1446  ......   0        1      189        0        0    189     1
2018-08-31 19:38:38.467   240.361 TCP     172.217.16.211:80    ->     192.168.10.2:57354 .AP.SF   0        5      560        0       18    112     2
2018-08-31 19:44:20.692     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:2059  ......   0        1      122        0        0    122     1
2018-08-31 20:05:09.724     0.000 UDP       192.168.10.2:62759 ->   208.67.222.222:53    ......   0        1       62        0        0     62     1
2018-08-31 19:38:47.967     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:42059 ......   0        1      226        0        0    226     1
2018-08-31 19:43:45.639     0.381 TCP       192.168.10.2:55457 ->      40.101.80.2:443   .AP.S.   0       10     1245       26    26141    124     1
2018-08-31 19:38:41.527   174.041 TCP       192.168.10.2:38929 ->   95.131.143.115:443   .APRS.   0       22     1954        0       89     88     2
2018-08-31 19:38:38.466   240.363 TCP     172.217.16.211:80    ->     192.168.10.2:57352 .AP.SF   0        5      560        0       18    112     2
2018-08-31 19:38:46.506  1580.698 TCP       192.168.10.2:34762 ->   64.233.166.108:993   .AP.S.   0       15     1844        0        9    122     2
2018-08-31 19:38:33.000     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:49467 ......   0        1       76        0        0     76     1
2018-08-31 20:05:09.003     0.000 UDP       192.168.10.2:56107 ->   208.67.222.222:53    ......   0        1       66        0        0     66     1
2018-08-31 19:31:00.001     0.000 TCP     34.248.128.178:5223  ->     192.168.10.2:36426 .AP...   0        1       97        0        0     97     1
2018-08-31 19:41:35.904    10.026 TCP     172.217.16.206:443   ->     192.168.10.2:56975 .A..SF   0        2      112        0       89     56     1
2018-08-31 19:38:32.747     0.000 UDP       192.168.10.2:34939 ->   208.67.220.220:53    ......   0        1       60        0        0     60     1
2018-08-31 19:38:33.015  1583.588 ICMP      192.168.10.2:0     ->     192.168.10.1:8.0   ......   0       14    14392        0       72   1028     5
2018-08-31 19:38:33.092     0.325 TCP        212.27.48.2:143   ->     192.168.10.2:47604 .AP.S.   0        9     1476       27    36332    164     1
2018-08-31 19:38:38.544     0.000 UDP     208.67.222.222:53    ->     192.168.10.2:35494 ......   0        1      267        0        0    267     1
2018-08-31 19:38:38.467   240.366 TCP     172.217.16.211:80    ->     192.168.10.2:57353 .AP.SF   0        5      560        0       18    112     2
2018-08-31 19:38:37.955   368.706 TCP       192.168.10.2:38111 ->   169.45.219.231:443   .APRS.   0       21     1501        0       32     71     4
2018-08-31 19:38:32.989     0.000 UDP       192.168.10.2:49467 ->   208.67.222.222:53    ......   0        1       60        0        0     60     1
2018-08-31 19:38:38.052   254.689 TCP     169.45.219.231:443   ->     192.168.10.2:38111 .AP.S.   0       17     1427        0       44     83     3
2018-08-31 19:38:41.542    10.263 TCP     95.131.143.115:443   ->     192.168.10.2:38930 .AP.SF   0       39    50791        3    39591   1302     1
[... cut ...]
2018-08-31 19:38:34.766     0.000 UDP       192.168.10.2:37859 ->   208.67.222.222:53    ......   0        1       56        0        0     56     1
2018-08-31 19:29:54.267   831.107 TCP       192.168.10.2:46263 ->     40.101.77.82:443   .AP.SF   0       13     1570        0       15    120     2
Summary: total flows: 204, total bytes: 590345, total packets: 1261, avg bps: 2202, avg pps: 0, avg bpp: 468
Time window: 2018-05-31 18:47:09 - 2018-08-31 20:05:17
Total flows processed: 204, Blocks skipped: 0, Bytes read: 12264
Sys: 0.366s flows/second: 556.4      Wall: 0.203s flows/second: 1004.2
```

Manifact, 2018-09-03:
Hello,

The `ps -ef | grep fprobe` command return : 

> pi        1405  1374  0 21:31 pts/0    00:00:00 grep --color=auto fprobe

Still `no matched flows` with the `nfdump -R /var/cache/nfdump -o extended -a` command on my side.

 I'd rather find another solution but do you think a complete re-install could solve the problem ?

Thank you.

pihomeserver, 2018-09-04:
Seems that fprobe process is not running
Can you please provide the output of `service fprobe status` ?

Try to start the service `service fprobe start` then redo the ps command to confirm that the service is running. If yes then use the hotspot to generate traffic and rerun tge nfdump command
If the service is still not running let's first try to run it manually with `/usr/sbin/fprobe -itun0 -fip 127.0.0.1:2055` in one terminal and then generate traffic and check the nfdump command

The complete install should confirm that the fprobe process is configured, started and running ... but should have been set in place at the first time.

Manifact, 2018-09-05:
Here is what the `service fprobe status` return : 
```
● fprobe.service - LSB: NetFlow Collector
   Loaded: loaded (/etc/init.d/fprobe; generated; vendor preset: enabled)
   Active: active (exited) since Mon 2018-09-03 19:37:54 UTC; 1 day 18h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 549 ExecStart=/etc/init.d/fprobe start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/fprobe.service

Sep 03 19:37:53 kazlab systemd[1]: Starting LSB: NetFlow Collector...
Sep 03 19:37:53 kazlab fprobe[549]: Starting fprobe: fprobe.
Sep 03 19:37:54 kazlab systemd[1]: Started LSB: NetFlow Collector.
```

When I try `service fprobe start`, I still have `pi        8307  8121  0 14:19 pts/0    00:00:00 grep --color=auto fprobe` with the command `ps -ef | grep fprobe`. 
When I try `/usr/sbin/fprobe -itun0 -fip 127.0.0.1:2055`, I now have : 
```
root      8298     1  0 14:18 ?        00:00:00 /usr/sbin/fprobe -itun0 -fip 127.0.0.1:2055
pi        8307  8121  0 14:19 pts/0    00:00:00 grep --color=auto fprobe
```

But still `no match flows` with traffic generated.

Manifact, 2018-09-05:
I just tried to reboot the raspberry : 
the command `ps -ef | grep fprobe `returns
`pi        1235  1199  0 14:27 pts/0    00:00:00 grep --color=auto fprobe`

But I now have the log which looks like it's working. 

```
Date first seen          Duration Proto      Src IP Addr:Port          Dst IP Addr:Port   Flags Tos  Packets    Bytes      pps      bps    Bpp Flows
2018-09-05 14:19:46.547    13.118 TCP       192.168.10.3:58271 ->       192.0.73.2:443   .A.RS.   0        3      140        0       85     46     1
2018-09-05 14:19:46.286     0.000 UDP       192.168.10.3:21199 ->   208.67.220.220:53    ......   0        1       53        0        0     53     1
2018-09-05 14:19:13.593    24.029 TCP       192.168.10.3:39084 ->    54.167.223.29:443   .A...F   0        2      104        0       34     52     1
2018-09-05 14:19:47.496    92.698 TCP       192.168.10.3:43126 ->    157.240.14.19:443   .AP..F   0      105     6114        1      527     58     2
2018-09-05 14:19:47.688     0.000 TCP       52.85.93.121:443   ->     192.168.10.3:37752 .A..S.   0        1       60        0        0     60     1
2018-09-05 14:19:12.297     0.001 TCP       192.168.10.3:58769 ->   34.236.164.173:443   .A...F   0        3      156     3000    1.2 M     52     1
2018-09-05 14:19:12.070     0.118 TCP      54.88.228.250:443   ->     192.168.10.3:33261 .AP..F   0        4      285       33    19322     71     1
2018-09-05 14:20:12.107   116.833 TCP     108.174.12.129:443   ->     192.168.10.3:39436 .AP.SF   0       12     2471        0      169    205     2
2018-09-05 14:19:37.908    46.989 TCP       192.168.10.3:39182 ->   213.246.58.109:443   .AP.S.   0      187    24299        3     4136    129     1
2018-09-05 14:19:38.115    43.385 TCP     213.246.58.109:443   ->     192.168.10.3:39181 .AP.SF   0      138   194995        3    35956   1413     1
2018-09-05 14:19:50.998   163.700 TCP       192.168.10.3:60023 ->   104.154.126.48:443   .AP...   0        5      282        0       13     56     2
2018-09-05 14:19:47.543    23.986 UDP       192.168.10.3:52760 ->   172.217.21.226:443   ......   0       20     5283        0     1762    264     1
2018-09-05 14:19:59.669    22.921 TCP       192.168.10.3:39221 ->   213.246.58.109:443   .AP.S.   0      117    20852        5     7277    178     1
2018-09-05 14:20:51.488     0.000 UDP       192.168.10.3:57903 ->   208.67.222.222:53    ......   0        1       60        0        0     60     1
2018-09-05 14:20:09.903     0.786 TCP       192.168.10.3:54929 ->      192.0.77.48:443   .AP.S.   0       11     1047       13    10656     95     1
2018-09-05 14:21:17.238     0.122 TCP     52.201.119.100:443   ->     192.168.10.3:53767 .AP..F   0        3      233       24    15278     77     1
2018-09-05 14:20:01.512     0.000 TCP      89.185.35.168:443   ->     192.168.10.3:34821 .A.R..   0        1       40        0        0     40     1
2018-09-05 14:20:15.463     0.219 TCP      104.79.255.73:443   ->     192.168.10.3:45405 .AP..F   0        3      187       13     6831     62     1
2018-09-05 14:19:37.622     0.000 TCP       192.168.10.3:39501 ->   104.79.252.145:443   .A...F   0        1       52        0        0     52     1
2018-09-05 14:19:20.925     5.001 TCP      34.235.48.210:443   ->     192.168.10.3:59752 .AP..F   0        2      135        0      215     67     1
2018-09-05 14:20:10.452    65.431 TCP      157.240.14.63:443   ->     192.168.10.3:52616 .APR.F   0       10     1607        0      196    160     2
2018-09-05 14:19:53.792     1.020 TCP     104.16.206.165:443   ->     192.168.10.3:46480 .AP...   0        4     3848        3    30180    962     1
2018-09-05 14:19:39.396    19.522 TCP       192.168.10.3:53139 ->   104.16.208.165:443   .AP.S.   0      130     9878        6     4047     75     1
2018-09-05 14:19:39.267    11.459 TCP       192.168.10.3:41421 ->   205.185.208.52:443   .APRS.   0       44     3607        3     2518     81     1
2018-09-05 14:19:39.493    11.200 TCP     54.231.176.181:443   ->     192.168.10.3:58606 .AP.S.   0       15     5795        1     4139    386     1
2018-09-05 14:19:39.137     0.000 UDP       192.168.10.3:8404  ->   208.67.222.222:53    ......   0        1       61        0        0     61     1
2018-09-05 14:19:46.558    13.108 TCP       192.168.10.3:58275 ->       192.0.73.2:443   .A.RS.   0        2      100        0       61     50     1
2018-09-05 14:19:39.561    13.472 UDP     172.217.21.226:443   ->     192.168.10.3:35783 ......   0       50    64892        3    38534   1297     1
2018-09-05 14:19:48.566   121.333 TCP       192.168.10.3:34820 ->    89.185.35.168:443   .AP...   0        9      834        0       54     92     2
2018-09-05 14:22:10.784     0.008 TCP       192.168.10.3:48450 ->   107.178.254.65:443   .A...F   0        2      116      250   116000     58     1
2018-09-05 14:22:06.585     0.420 TCP     35.186.251.138:443   ->     192.168.10.3:42438 .AP..F   0        4      271        9     5161     67     1
2018-09-05 14:18:58.242     0.000 TCP       192.168.10.3:52494 ->    157.240.14.63:443   .A...F   0        2      104        0        0     52     1
2018-09-05 14:20:12.011    65.672 TCP      157.240.14.15:443   ->     192.168.10.3:52180 .AP.SF   0       14     4650        0      566    332     2
2018-09-05 14:18:58.545     0.000 UDP     208.67.222.222:53    ->     192.168.10.3:27332 ......   0        1       98        0        0     98     1
[... cut ...]
2018-09-05 14:21:41.141     0.000 TCP       192.168.10.3:45621 ->   172.217.22.110:443   .AP...   0        1       83        0        0     83     1
2018-09-05 14:20:11.775     0.000 UDP       192.168.10.3:37929 ->   208.67.222.222:53    ......   0        1       62        0        0     62     1
Summary: total flows: 286, total bytes: 3177812, total packets: 4384, avg bps: 101443, avg pps: 17, avg bpp: 724
Time window: 2018-08-27 18:20:54 - 2018-09-05 14:23:08
Total flows processed: 286, Blocks skipped: 0, Bytes read: 20576
Sys: 0.032s flows/second: 8799.2     Wall: 0.018s flows/second: 15468.7

```

Thank you !

pihomeserver, 2018-09-05:
Good news ! Seems to come from the probe service that did not start as expected
I will develop within the web interface a log browser

Also keep in mind that nfdump updates logs every 5 minutes. It's not real time

closed 2018-09-05 by pihomeserver

Photonoxx, 2019-08-30:
Hi,

I know this file is closed, but I would like to report I had the same issue (fprobe don't start due to tun0 not available at boot time) and adding in /etc/init.d/fprobe script a "sleep 5" in the beginning of start part of the script seems to fix the matter. After that, I'm not sure modify Init.d file is a perfect thing to do, but it could help people with the same matter.

pihomeserver, 2019-08-30:
The startup of services should be reviewed to rewrite in SystemV and add dependencies. So then wait for Coova to start before starting fprobe