pihomeserver/Kupiki-Hotspot-Script #129: Modify the captive portal page [Question]

joyer0, 2018-09-12:
Hey !

How do you change what is displayed on the captive portal page ? 
I mean the text and the terms for example ? 

Also i didn't manage to see the connection logs, with the MAC address. Where are they ?

Thanks :)

pihomeserver, 2018-09-12:
Bonjour

1. Il faut changer le fichier /usr/share/nginx/portal/js/configuration.json
2. Les logs ne sont pas encore accessibles directement. Voir #127 pour les commandes

pihomeserver, 2018-09-12:
Pour la case c'est le css qui fait cet effet. Possible de le changer en changeant le CSS (autour de la ligne 280 dans portal.css)

Mauvaise gestion de l'erreur effectivement. JE dois regarder ça

joyer0, 2018-09-12:
Super merci ! Excellent travail en tout cas 😉