pihomeserver/Kupiki-Hotspot-Script #130: How do I change the captive portal? Want to Word press website

acujl, 2018-09-18:
After connect to the Wifi a default captive portal shows up, want to customize it to a wordpress website. I don't need the features of users, etc... just want to the website show up first, ant then the user can navigate the web..

#### Configuration parameters (if changed in the script)

# Name of the log file
LOGNAME="kupiki_hotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="Free Wifi"
# IP of the hotspot
HOTSPOT_IP="192.168.10.1"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wc$
WIFI_COUNTRY_CODE="PT"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.10.0"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head$
# WAN interface (the one with Internet - default 'eth0' or long name for Debian$
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$$
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="nl80211"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="N"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="Y"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed chec$
# Set value to Y or N
NETFLOW_ENABLED="Y"

### Your environment

- Hardware : Raspberry Pi 3 using the integrated wifi
- Operating system: raspian

pihomeserver, 2018-09-19:
You can link the portal URL in Nginx configuration to whatever you want (in local only of course) 

For the automatic authentication, i don't see how to do that if you connect a wordpress and the browse. 
Usually the captive portal is called by the device when you connect to the hotspot. You can use a dummy button that will use a defined user set in freeradius. Then you will always authenticate.
Another approach could be an iptable in the up.sh of coova that will open the internet access as soon as Coova is up. So no need of authentication

acujl, 2018-09-19:
How can I do that..? I'm used to set up apache webservers..

pihomeserver, 2018-09-19:
1) Install Wordpress in the way you want
2) Copy the URL you want your portal goes to
3) Paste the URLin the /etc/chilli/defaults in the parameter HS_UAMFORMAT
4) Stop/start the chilli service

acujl, 2018-09-20:
Btw to anyone who is reading this, to access mysql:
sudo -i
mysql -p
passowrd: pihotspot

acujl, 2018-09-20:
And if you want to install wordpress website just create database, user, etc with mysql
Then go to /usr/share/nginx/portal and make your installation there
Also go to /etc/nginx/sites-available/portal and it should be like this:

```
server {
        listen 192.168.10.1:80 default_server;

        root /usr/share/nginx/portal;

        index index.php index.html index.htm index.nginx-debian.html;

        server_name _;

        location / {
                try_files $uri $uri/ =404;
        }

        location ~ \.php$ {
                include snippets/fastcgi-php.conf;
                fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
        }
}
```
other wise it won't work

DSBTechSoftware, 2018-10-16:
Hi can someone point me in the right direction, i want to display a local page after a successful login, ?

pihomeserver, 2018-10-16:
@DSBTechSoftware that's another issue. Replace the line 112 of js/portal.js with the action to redirect to the expected location. Keep in mind that you will see the page only few seconds as it's usually closed by the system when internet is reachable (at least on phones)

closed 2018-11-07 by pihomeserver