pihomeserver/Kupiki-Hotspot-Script #133: ./pihotspot.sh: line 532: mariadb: command not found

jgamos, 2018-09-24:
### Actual behavior

command not found

### Expected behavior

line 532: mariadb:

#### Configuration parameters (if changed in the script)

./pihotspot.sh: line 532: mariadb: command not found

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : RPI 3
- Operating system: Raspbian Stretch

AGUERY, 2018-09-24:
Hello 
I have the same issue on a rasppberry 3B+ running install as root 
What could I check ?

pihomeserver, 2018-09-25:
I will check that, maybe an issue with a new release package of mariadb

@AGUERY : Can you please have a look in the /var/log folder and check the content of pihomeserver file ?

pihomeserver, 2018-09-25:
As a workaround, please set INSTALL_KUPIKI_ADMIN parameter to N
That will not install the new frontend

AGUERY, 2018-09-27:
Hello,
I have just tried to re install but Always an issue.
parameter  INSTALL_KUPIKI_ADMIN =N
Below the message diplayed/

" Your system is up to date! Continuing with Pi-Hotspot installation...
:::
::: Removing any previous sources of CoovaChilli project
::: Cloning CoovaChilli project
::: Cloning daloradius project
::: Cloning Pi Hotspot portal project
::: Starting MySql service
:::
::: Something went wrong with the last command. Please check the log file"

AGUERY, 2018-09-27:
Here is the content of kupiki_hotspot.log

Cloning into 'coova-chilli'...
::: Cloning daloradius project
Cloning into 'daloradius'...
::: Cloning Pi Hotspot portal project
Cloning into 'portal'...
::: Starting MySql service
Failed to restart mariadb.service: Unit mariadb.service not found.
:::
::: Something went wrong with the last command. Please check the log file

12e1121, 2018-09-28:
It seems that you stuck in mariadb during installation. I had this issue before but i solved it by manually install mariadb. You may try the following:

sudo apt-get install -f
sudo apt-get install mariadb-server-10.1

pihomeserver, 2018-09-28:
Thanks @12e1121 
I will try again the installer this week end, maybe the package name has changed as i use mariadb-server without the version number

AGUERY, 2018-10-01:
Hello,
It is better now but always an error

root@raspberrypi:~# tail -f /var/log/kupiki_hotspot.log
::: Updating nfdump service configuration (systemd)
::: Set cleaning options for Netflow data
::: Adding logs cleaning every day at 1am (system time)
::: Adding Netflow stats update in the crontab
00 01 * * * root /usr/bin/nfexpire -r /var/cache/nfdump
::: Stopping freeradius service to update the configuration
Failed to stop freeradius.service: Unit freeradius.service not loaded.
:::

pihomeserver, 2018-10-02:
Please use the latest release of the script. I tested it on last friday and it works like a charm with default parameters (no issue with mariadb). It could require a reboot and running the script again due to kernel modules to load

AGUERY, 2018-10-02:
I have now an issue with the coovachili building.

pihomeserver, 2018-10-02:
Hummm .... Let's start from scratch : fresh Raspbian **Lite** (that's important) install with latest release of the script ?
If so then please share the output to help to investigate

AGUERY, 2018-10-02:
I think m'y problem is I am using omv4 distro

pihomeserver, 2018-10-02:
Pretty sure that's the issue

closed 2018-10-04 by pihomeserver

pihomeserver, 2018-11-21:
@jgamos impossible to help with so few information !