pihomeserver/Kupiki-Hotspot-Script #134: MultiHotSpot

12e1121, 2018-09-27:
Hi, thanks for the awesome project! I did follow up with your tutorial and script to build the HotSpot and everything is working fine. Now i would have question regarding on multiple HotSpots. My setups:

HotSpot A (freeradius Server A):
-freeradius server enabled
-chilli enabled
-nginx enabled

HotSpot B:
-freeradius server disabled
-chilli enabled
-nginx enabled

HotSpot C:
-freeradius server disabled
-chilli enabled
-nginx enabled

All HotSpots having same:
-WiFi interface, SSID, and Channel.
-Hardware : RPI 3
-Operating system: Raspbian Stretch

I am able to connect HotSpot B and HotSpot C to freeradius Server HotSpot A, by configuring _clients.conf_ referencing #113 . (Meaning that i can use users that registered from HotSpot A). 

According to https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/43#issuecomment-314054964, how can i roaming from one HotSpot to another without relogin?

I did have a dirty hack for this solution. _Since all HostSpots generating different sessions in **radacct** table, we would just need to insert same session for different  stationid(HotSpots)_. 

Do we have any other proper solution for this?

pihomeserver, 2018-10-01:
Hi. Late reply ... I don't think it's possible as when you go on another hotspot, the name is the same but it's another wifi network so that requires a new authentication with user/password. 
Maybe with the MAC address it could work as it's automatic. To be checked

12e1121, 2018-10-02:
Thanks. I will try to check with MAC address authentication then.

EasyVoiceTel, 2018-10-03:
I'm pretty curious at this one! I just use multiple pi's and  add users manually, maybe a sync function is possible for an update?

pihomeserver, 2018-10-03:
@EasyVoiceTel  that's not required you have to use only one instance of Freeradius

EasyVoiceTel, 2018-10-03:
> @EasyVoiceTel that's not required you have to use only one instance of Freeradius

Really how to tie them together?

12e1121, 2018-10-04:
@EasyVoiceTel It is quite simple to link them together actually. You may referencing  #113.

EasyVoiceTel, 2018-10-09:
> @EasyVoiceTel It is quite simple to link them together actually. You may referencing #113.

Okay I read  the topic but does the pie's need to see each other via wifi?

pihomeserver, 2018-10-09:
No. The point is that you just share the same freeradius server.

EasyVoiceTel, 2018-10-09:
Like how is there an a to z manual for that? Thanks (still a newbie)

pihomeserver, 2018-10-09:
Look at the original post. On "slave" pi in the coova configuration set the param for the radius ip server with the ip of the master one. Of course it must be reachable !

reigelgallarde, 2019-04-18:
any reason why you would want this kind of setup @12e1121 ?

pihomeserver, 2019-04-18:
@reigelgallarde That could be useful to have only central place to declare, store and mange users for different access points

12e1121, 2019-04-18:
@pihomeserver  Indeed. However, I seems can't make sure every connections / packets from multiple access points route to central place.

reigelgallarde, 2019-04-18:
Anyway, I asked because this might not be the ideal setup.

Like, why not use some outdoor AP that has a wider range? Because I've read his setup has the same WiFi interface, SSID, and Channel... Thinking, if the hotspot is just in one area, wider range AP should be suitable solution...

12e1121, 2019-04-18:
@reigelgallarde I think it is a different scenario. My intention is to **increase the number of APs** and one Master AP control of multiple salves AP instead of **improve the range of AP**.