pihomeserver/Kupiki-Hotspot-Script #135: User doesn't get a notification when the raspberry is not connected to the internet

Daniel-Bertagnolli, 2018-09-28:
### Actual behavior

We are using the hotspot to show some internal pages to the logged in users but if the raspberry is not connected to internet the users don't get a notification "you need to access to use this network" (some phones don't show it even when it's connected, for example my honor 9). If then the user tries to browse the internet there is no automatic redirect to the login page, for both HTTP and HTTPS requests, they'll have to type in the direct IP to see the login page.

### Expected behavior

Every time a user connects to the wireless network he gets the notification where he needs to login to use the network, and if he tries to browse other websites before logging in he gets redirected to the login page

### Your environment

- Hardware : raspberry pi3
- Operating system: raspbian (latest version)

### Additional comments

The system, when in a production environment will NOT be connected to the internet.

pihomeserver, 2018-09-28:
From what i understand, it's that without Internet connection on the Raspberry Pi, the portal does not pop up ? Never tested that but that sounds to be an interesting feature
I will look at that. Share your progress in case of you try to investiguate
Thanks

Daniel-Bertagnolli, 2018-10-01:
Exactly! The portal pops up ONLY if i reach it with it's direct IP, in our case 192.168.10.1 

I found a workaroud (it's pretty brutal as a workaround) and it works with another hotspot script, I'll gladly try with kupiki as soon as i can and then i'll tell you if it works and how to configure it

pihomeserver, 2018-11-07:
@Daniel-Bertagnolli any feedback to share about that issue ?

Daniel-Bertagnolli, 2018-11-28:
Hi, sorry for not answering sooner! 

So, we tried this workaround yesterday and sadly it didn't work. Mostly because to work we have to run a few commands, then kupiki and then a add few firewall rules but apparently kupiki starts as soon as the machine boots up so there is not chance for us to run commands before kupiki launches. If there is a way to manually launch kupiki instead of launching automatically at system boot maybe it might work :)

pihomeserver, 2018-11-29:
DEpends what you call Kupiki ? It's only a ready to use configuration of hostapd, coova, freeradius and mariadb.
Then identy before which tool your script should be executed and then create a service in init.d with dependency. You should then master the start sequence

ziedbh, 2018-12-11:
Hello, I have an issue detected only on samsung s4 and j7 but ok on others (ios, lenovo, asus, huawei)
I installed a webserver with captive portal on a raspberry with a hotspot (nginx/apache2/lighttpd, hostapd, dnsmasq) when i connect to this wifi network a popup "Sign-in to network" is displayed or an auto launch of the captive portal page is done.
This behavior failed only on some samsung smartphone like s4 and j7, when i connect nothing happened!!! no popup nor auto display CP page
any proposition?

pihomeserver, 2018-12-12:
@ziedbh : no need to create multiple message in issues to get help. You opened one that's enought.