pihomeserver/Kupiki-Hotspot-Script #136: Login page Jason connection lost error

EasyVoiceTel, 2018-09-29:
### Actual behavior

According to earlier responds on this form; 
installation went fine, but wen I connect to my hotspot the login page pops an error about Jason who is losing the connection  see image:

[image]

Can it be that it has some thing to do with dhcpcd? Or am I talking nonsense now? I'm still a noob to raspberry!

cause thats the only thing that says failed during startup. 

### Expected behavior

No issues 

#### Configuration parameters (if changed in the script)

#!/bin/bash

# PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION

# Name of the log file
LOGNAME="kupiki_hotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="pihotspot"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="EasyVoice-Telecom-Hotspot"
# IP of the hotspot
HOTSPOT_IP="192.168.10.1"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="FR"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="Y"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.10.0"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
# WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="nl80211"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="N"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="N"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
# Set value to Y or N
NETFLOW_ENABLED="Y"
[... 1017 lines cut ...]
display_message "- Wifi Hotspot available : $HOTSPOT_NAME"
if [ $AVAHI_INSTALL = "Y" ]; then
    display_message "- For the user management, please connect to http://$MY_IP/ or http://$HOTSPOT_NAME.local/"
else
    display_message "- For the user management, please connect to http://$MY_IP/"
fi
display_message "  (login : administrator / password : radius)"

exit 0

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment
Raspberry Pi model B + 
Raspbian lite! 

### Additional comments

{Please write here, if there is something more to tell}

EasyVoiceTel, 2018-09-30:
I decided to go over the tutorial again, with a fresh flash of raspbian lite.. 

I do not change any setting. tell the results later on.

EasyVoiceTel, 2018-09-30:
now it even don't get passed the installation it hangs at reloading ssh services. I'll contact you by e-mail to discuss something else!

EasyVoiceTel, 2018-09-30:
dhcpcd is failling to start even manual I checked it but it is default as installed with the script!

pihomeserver, 2018-10-01:
Yes, that's not required by the system
Please provide the output of the command `ifconfig -a` executed with root user

EasyVoiceTel, 2018-10-01:
Uh, I don't really know what happend but suddenly it came to live... 

[image]

closed 2018-10-01 by EasyVoiceTel

EasyVoiceTel, 2018-10-04:
Hi there, there could be many reasons for it to not work, in my case I guess that the SD card was faulty, that’s why I strongly recommend using an sdXC