pihomeserver/Kupiki-Hotspot-Script #138: open for write: permission denied on editing logo, background and js file!

EasyVoiceTel, 2018-10-02:
### Actual behavior
I try to change the logo, background and the js file used for the terms and conditions, but it gives me this error using filezilla. 

/usr/share/nginx/portal/images/logo.png: open for write: permission denied

### Expected behavior

Just be able to copy the files

#### Configuration parameters (if changed in the script)

Hostname only
Opdracht: 	cd "/usr/share/nginx/portal/images"
Antwoord: 	New directory is: "/usr/share/nginx/portal/images"
Opdracht: 	put "/Users/MADE UNKNOWN DUE TO PRIVACY/desktop/pi hotspot/logo.png" "logo.png"
Fout:         	/usr/share/nginx/portal/images/logo.png: open for write: permission denied
Fout:         	Bestandsoverdracht mislukt

pihomeserver, 2018-10-02:
That's not related to the script but basic rights management. Look at the file on the SD card and be sure to use the same user from Filezilla.

EasyVoiceTel, 2018-10-02:
What do you mean by look at the file on the SD card? Im using filezilla with the username pi over ssh the same way I normally connect to throw commands to it with terminal or terminus on my mobile device.

Are there any commands which let me take ownership of those files?

pihomeserver, 2018-10-02:
The files you want to change are not owned by the pi user for security reasons. I don't have in mind the default value. Go to `/usr/share/nginx/portal`, list files, look at the owner or change right to get full access even if not the best way to do (`chmod -R 777 *`)

EasyVoiceTel, 2018-10-02:
sudo chmod -R 777 /usr/share/nginx/portal did the trick. 
Please don't forget to include sudo!

closed 2018-10-02 by EasyVoiceTel

EasyVoiceTel, 2018-10-08:
Thanks to your help I managed to do this 👍 🥇 

[image]
[image]