pihomeserver/Kupiki-Hotspot-Script #139: For all who don't want to hassle with the setup! Here is an .img just flash it and go!

EasyVoiceTel, 2018-10-03:
Hi there, 

Running in to errors, or simply don't have the time to hassle with the installation?! 
Here is my img file: https://easycompzeeland.nl/downloads/easyvoice-hotspot.img.zip (download available on 00:00 +2 Amsterdam) due to the fact that it's still uploading! 

I used a San Disk 16GB SDXC card! I strongly recommend using a simillar card! 
Img created using Apple-PI-baker for MacOS tested flashing with Etcher! (cause it's better)

All passwords such as root, pi and the management panel are default! 
Setting changed from setup; Hostname. 
You can change it by re running the setup (but then downloading this img would be useless🤯) 

That's all, and remember: One Wish 🌠 And A Very Big Fish 🐡!

EasyVoiceTel, 2018-10-03:
For support refer to issues I just created an img with my installation! Other messages on this topic are fine 👍

rubenmdh, 2018-11-04:
Hello,
how is it possible that uncompressed uses 15,93GiB? Are you including the whole partition?

My installation (completely fresh Raspbian Lite with Kupiki installed) uses 2GiB, which makes way more sense. I extracted this information from _df -h_.

pihomeserver, 2018-11-05:
The problem is that the image is the size of the SD card (including free space). That's why i did not already made one final image (also because of the re configuration once all is installed that would generate too many git issues tickets !)
What i would prefer is a minimal partition that would auto expand at first run.

EasyVoiceTel, 2018-11-05:
> Hello,
> how is it possible that uncompressed uses 15,93GiB? Are you including the whole partition?
> 
> My installation (completely fresh Raspbian Lite with Kupiki installed) uses 2GiB, which makes way more sense. I extracted this information from _df -h_.

The first part of @pihomeserver is correct. it's the complete SD card which I used including free space so I recommend using 15 GB card. However the second part is a bit untrue cause i used my own partition for a several end-clients which didn't gave any issues what so ever. its just plug and play.

oxgx, 2019-03-20:
..Pi-Hotspot $ sudo ./pihotspot.sh
...
::: Updating freeradius default configuration (2)
::: Activating COA support in Freeradius
::: Checking freeradius configuration
::: Activating IP forwarding
::: Restarting network service to take IP forwarding into account
::: Building CoovaChilli package <<<<<<<<<<<<<<<<<<<<<<< this is where it gets stuck all the time

Any idea?

pihomeserver, 2019-03-20:
Use the script on a fresh image not the one in that post

oxgx, 2019-03-20:
Not sure what pots do you mean. I've installed EasyVoiceTel image,plugged it in, updated it and hit the script.

pihomeserver, 2019-03-20:
I don't know how the image has been made and if it's updatable. Follow instructions from the README to install all from scratch. On the image the script seems to have been already executed (an old version of it).

stewert69, 2020-01-28:
Trying to download image gives a 404 error.

EasyVoiceTel, 2020-01-28:
There has been a change on our host, I’ll try to re upload it soon! 
Thanks for notifying me.

manish-448, 2020-08-29:
hi
showing 

Page not found
The page you are looking for was not found.

Can you please send a new link 

thanks in advance

chivarpilones, 2022-04-07:
> Hi there,
> 
> Running in to errors, or simply don't have the time to hassle with the installation?! Here is my img file;
> 
> https://easycompzeeland.nl/downloads/easyvoice-hotspot.img.zip (Compressed file size 3.23 GB | Uncompressed 15,93 GB)
> 
> I used a San Disk 16GB SDXC card! I strongly recommend using a simillar card! Img created using Apple-PI-baker for MacOS tested flashing with Etcher! (cause it's better)
> 
> If you experience WAN issues? Just run the install script, this will check the IP.
> 
> All passwords such as root, pi and the management panel are default! Setting changed from setup; Hostname. You can change it by re running the setup (but then downloading this img would be useless🤯)
> 
> That's all, and remember: One Wish 🌠 And A Very Big Fish 🐡!

Hi, Do you have an updated link for this? like in gdrive or mediafire or mega?