pihomeserver/Kupiki-Hotspot-Script #140: Can it be used on a Hot banana PI?!

EasyVoiceTel, 2018-10-03:
Can it be used on a Banana PI? like this one; 
[image]

Found it on;
https://nl.aliexpress.com/item/Hot-Banana-Pi-R1-Wireless-smart-Router-BPI-R1-case-Acrylic-Enclosure-2pcs-BPI-R1/32256050317.html?spm=a2g0z.10010108.1000016.1.2f5a7098krAXpx&isOrigTitle=true

rubenmdh, 2018-11-04:
Hello,
while I haven't tried on a Banana Pi, I've successfully installed my Kupiki hotspot on a Raspberry Pi 2 with a Realtek USB dongle. (Whereas the recommended is the Pi 3 and up)

I guess your best bet is to try it by yourself if you already have the Banana Pi, otherwise I would stick with the RPi for convenience.

Just make sure you install Raspbian Lite and change the Wi-Fi adapter driver in the script.

pihomeserver, 2018-11-05:
I had an old version and it was like a Raspberry. If it's compatible Debian then Kupiki script should be run with some parameters updates

closed 2019-02-13 by pihomeserver