pihomeserver/Kupiki-Hotspot-Script #142: Wlan0 - > hot-spot and wlan1 - > internet connection

acujl, 2018-10-25:


pihomeserver, 2018-10-28:
Hello
Did you get any error message other than that ? What is the output of the list of interfaces ?

acujl, 2018-11-01:
I went and installed everything again....
Before instalation the interfaces looked nominal, I could acess wifi on wlan0 and wlan1 (my adapter) etc.

During the instalation I had an error yes..
```

::: Activating the banner for SSH
:::
::: Creating Kupiki Admin folder for the database
::: Changing rights of the folder
::: Updating authentication plugin
::: Starting freeradius service
:::
::: Something went wrong with the last command. Please check the log file
:::
```
After this interfaces just died including eth0, I had to acess it remotly 
[image]

[kupiki_hotspot.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/2538153/kupiki_hotspot.log)

pihomeserver, 2018-11-04:
1) Did you changed the hotspot driver for your wifi interface if you don't use the built in adaptater ? There is a parameter for that in the script
2) Freeradius is not able to start. I guess the first steps should be
- Check if hostapd is able to start. Look if the process is running 
- If it runs then start coova using command `coova -f -d` (to be confirmed)
- If it runs then start freeradius using command `freeradius -X`

closed 2019-02-13 by pihomeserver