pihomeserver/Kupiki-Hotspot-Script #143: Installer fails if SSH is not running

rubenmdh, 2018-11-04:
Hello,
I am trying to install the Kupiki hotspot in my RPi 2 and I found an annoying error which could be easily solved beforehand.

The thing is that I installed the latest Raspbian Lite image and by default it doesn't come with SSH enabled. (and therefore, it is not running)

Well, after waiting for a few minutes for the CoovaChilli package to build, the installer stopped because it couldn't restart the SSH daemon as it was not running. I enabled it from raspi-config and everything went fine, but this is kinda annoying.

I think there should be an initial check to see if the SSH daemon is running or not (to avoid having to reinstall Kupiki again) or, at least, indicating somewhere in the readme that SSH is strictly needed for the script to continue.

Thank you very much for this script.

pihomeserver, 2018-11-05:
Hello
Good point. I restart the service to take into account the login message. I will perform a test before updating the login message

pihomeserver, 2018-11-06:
Corrected in v2.0.6 script release

closed 2018-11-06 by pihomeserver