pihomeserver/Kupiki-Hotspot-Script #144: Login JSON error

rubenmdh, 2018-11-04:
Hello,

Login fails with the error on the client: "The last contact with the controller (10.2.5.1:3990) failed
Error: JSON request timed out (or reply is not valid)"

I created beforehand a user with name and clear-password "test"

I have searched for it before and the only thing I've found useful is this thing when running **freeradius -X**:

```
(0) Received Access-Request Id 16 from 127.0.0.1:57242 to 127.0.0.1:1812 length 299
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(0) Cleaning up request packet ID 16 with timestamp +18
Ready to process requests
```

Hope you can help me.

pihomeserver, 2018-11-05:
Can you please provide expected information for any new issue ? It's usually a misconfiguration of the parameters in the script of that coova has not been started

rubenmdh, 2018-11-05:
Hello, I guess you are referring to the pihotspot script. Here it is:
```
#!/bin/bash

# PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION

# Name of the log file
LOGNAME="kupiki_hotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="f6YDmbEVDp8UuntK8RKvY9lhpnAHpP4"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="TestHotspot2"
# IP of the hotspot
HOTSPOT_IP="10.2.5.1"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="10.2.5.0"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
# WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="8192cu"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="Y"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="Y"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
# Set value to Y or N
NETFLOW_ENABLED="Y"
# Define how long Netflow logs will be stored
# Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
# Accepted time scales are w (weeks) d (days) H (hours).
# A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
# By default data are stored 365 days (value set to 365d)
ETFLOW_LOGS_DAYS="365d"
# Enable/Disable MAC authentication
# Set value to Y or N
MAC_AUTHENTICATION_ENABLED="N"
# Password for MAC authentication. Could/Should be changed within the web administration interface
MAC_AUTHENTICATION_PASSWORD="123456"
# Install web frontend of Kupiki Hotspot
# Set value to Y or N
INSTALL_KUPIKI_ADMIN=Y
# Install Cron job for the hotspot updater. Will be executed every sunday at 8am (system time)
# Set value to Y or N
ADD_CRON_UPDATER=Y
```

I've tried with several configurations (in the latest one I enabled Haserl). The LAN_WIFI_DRIVER doesn't matter because whether I put 8192cu or rtl8192cu I always have to comment out the line on hostapd.conf to make the hotspot visible.

rubenmdh, 2018-11-05:
Hello, I guess you are referring to the pihotspot script. Here it is:
```
#!/bin/bash

# PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION

# Name of the log file
LOGNAME="kupiki_hotspot.log"
# Path where the logfile will be stored
# be sure to add a / at the end of the path
LOGPATH="/var/log/"
# Password for user root (MySql/MariaDB not system)
MYSQL_PASSWORD="[REDACTED]"
# Name of the hotspot that will be visible for users/customers
HOTSPOT_NAME="TestHotspot2"
# IP of the hotspot
HOTSPOT_IP="10.2.5.1"
# Wi-fi code country. Use above link to find yours
# https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
WIFI_COUNTRY_CODE="ES"
# Use HTTPS to connect to web portal
# Set value to Y or N
HOTSPOT_HTTPS="N"
# Network where the hotspot is located
HOTSPOT_NETWORK="10.2.5.0"
# Secret word for FreeRadius
FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
# WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
# LAN interface (the one for the hotspot)
LAN_INTERFACE="wlan0"
# Wifi driver
LAN_WIFI_DRIVER="8192cu"
# Install Haserl (required if you want to use the default Coova Portal)
# Set value to Y or N
HASERL_INSTALL="Y"
# Password used for the generation of the certificate
CERT_PASSWORD="pihotspot"
# Number of days to certify the certificate for (default 2 years)
CERT_DAYS="730"
# Make Avahi optional
# Set value to Y or N
AVAHI_INSTALL="Y"
# Install Daloradius Portal (compatible with FR2 only in theory)
# Set value to Y or N
DALORADIUS_INSTALL="Y"
# Enable/Disable Bluetooth
# Set value to Y or N
BLUETOOTH_ENABLED="N"
# Enable/Disable fail2ban to protect server from unwanted access
# Set value to Y or N
FAIL2BAN_ENABLED="Y"
# Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
# Set value to Y or N
NETFLOW_ENABLED="Y"
# Define how long Netflow logs will be stored
# Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
# Accepted time scales are w (weeks) d (days) H (hours).
# A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
# By default data are stored 365 days (value set to 365d)
ETFLOW_LOGS_DAYS="365d"
# Enable/Disable MAC authentication
# Set value to Y or N
MAC_AUTHENTICATION_ENABLED="N"
# Password for MAC authentication. Could/Should be changed within the web administration interface
MAC_AUTHENTICATION_PASSWORD="123456"
# Install web frontend of Kupiki Hotspot
# Set value to Y or N
INSTALL_KUPIKI_ADMIN=Y
# Install Cron job for the hotspot updater. Will be executed every sunday at 8am (system time)
# Set value to Y or N
ADD_CRON_UPDATER=Y
```

I've tried with several configurations (in the latest one I enabled Haserl). The LAN_WIFI_DRIVER doesn't matter because whether I put 8192cu or rtl8192cu I always have to comment out the line on hostapd.conf to make the hotspot visible.

pihomeserver, 2018-11-05:
Thank you
Do you confirm that your WAN IP is different than 10.2.5.0 ? Did you also check if Coova service is up and running ? You can start it in debug mode is needed. This error comes from coova service not reachable
Also using command `ifconfig -a` should list all network interfaces including a tun0 that is created by coova if it's running

rubenmdh, 2018-11-05:
Hello, my WAN IP is on the 192.168.5.0 subnet and I can confirm that the tun0 interface is showing up in `ifconfig -a`.

I've run chilli with the debug option and it's really weird (I don't understand why it doesn't work):

```
pi@kupikihotspot:~ $ sudo chilli --debug --fg
chilli[1271]: (Re)processing options [/var/run/chilli.1271.cfg.bin]
chilli[1272]: running chilli_opt on /var/run/chilli.1271.cfg.bin
chilli[1271]: PID 1271 rereading binary file /var/run/chilli.1271.cfg.bin
chilli[1271]: Loading modules
chilli[1271]: PID 1271 reloaded binary options file
chilli[1271]: CoovaChilli 1.4. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-2012 David Bird (Coova Technologies). Licensed under GPL. See http://coova.github.io/ for details.
chilli[1271]: chilli_main(7416): clock realtime sec 1541410361 nsec 451022355
chilli[1271]: chilli_main(7423): clock monotonic sec 203 nsec 217280497
chilli[1271]: TX queue length set to 100
chilli[1271]: Running /etc/chilli/up.sh
chilli[1271]: Hashlog 8 253 256
chilli[1271]: Net SNDBUF 163840
chilli[1271]: Net RCVBUF 163840
chilli[1271]: device wlan0 ifindex 3
chilli[1271]: lo  address family: 17 (AF_PACKET)
chilli[1271]: eth0  address family: 17 (AF_PACKET)
chilli[1271]: wlan0  address family: 17 (AF_PACKET)
chilli[1271]: lo  address family: 2 (AF_INET)
chilli[1271]: eth0  address family: 2 (AF_INET)
chilli[1271]: tun0  address family: 2 (AF_INET)
chilli[1271]: dhcp_hashinit(490): hash table size 64 (56)
chilli[1271]: dhcp_sendGARP(338): GARP: Replying to broadcast
chilli[1271]: dhcp_new(1191): dhcpif (wlan0) IPv6 address ::
chilli[1271]: RADIUS client 0.0.0.0:3799
chilli[1271]: chilli_main(7579): Waiting for client request...
chilli[1271]: net select count: 1
chilli[1271]: net select count: 2
chilli[1271]: net select count: 3
chilli[1271]: net select count: 4
chilli[1271]: net select count: 5
chilli[1271]: net select count: 6
chilli[1271]: net select count: 7
chilli[1271]: i=0 sz=1
chilli[1271]: i=0 sz=2
chilli[1271]: i=1 sz=2

System Interfaces
0) lo (1) ip=127.0.0.1 net=127.0.0.0 mask=255.0.0.0 mac=00-00-00-00-00-00 mtu=65536
1) eth0 (2) ip=192.168.5.109 net=192.168.5.0 mask=255.255.255.0 bcase=192.168.5.255 mac=[REDACTED]-a7 mtu=1500
2) tun0 (5) ip=192.168.10.1 net=192.168.10.0 mask=255.255.255.0 peer=192.168.10.1 mac=00-00-00-00-00-00 mtu=1500

System Routes
0) dst=0.0.0.0 mask=0.0.0.0 gw=192.168.5.1 dev=eth0 (2)
1) dst=192.168.5.0 mask=255.255.255.0 dev=eth0 (2)
2) dst=192.168.10.0 mask=255.255.255.0 dev=tun0 (5)
chilli[1271]: Default Route 192.168.5.1
chilli[1271]: Route Interface eth0
chilli[1271]: MAC [REDACTED]:fd
chilli[1271]: net select count: 8
chilli[1271]: net select count: 9
chilli[1271]: dhcp_receive_ip(3771): Address not found (192.168.5.2)
chilli[1271]: dhcp_newconn(745): DHCP newconn: [REDACTED]-67
chilli[1271]: cb_dhcp_connect(5222): New DHCP request from MAC=[REDACTED]-67
chilli[1271]: chilli_connect(5183): New Chilli Connection
chilli[1271]: dhcp_receive_ip(3927): Not for our MAC or broadcast: [REDACTED]-A7
chilli[1271]: dhcp_receive_ip(3771): Address not found (192.168.5.109)
chilli[1271]: dhcp_newconn(745): DHCP newconn: [REDACTED]-A7
chilli[1271]: cb_dhcp_connect(5222): New DHCP request from MAC=[REDACTED]-A7
chilli[1271]: chilli_connect(5183): New Chilli Connection
chilli[1271]: dhcp_receive_ip(3927): Not for our MAC or broadcast: [REDACTED]-67
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3927): Not for our MAC or broadcast: [REDACTED]-67
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3927): Not for our MAC or broadcast: [REDACTED]-A7
[... 193 lines cut ...]
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1337]: Rescheduling RADIUS request id=7 idx=0
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_arp(5954): ARP: [REDACTED]-AA asking about 192.168.10.1
chilli[1271]: dhcp_sendARP(5829): ARP: Replying to 192.168.10.2 / [REDACTED]-AA
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dns_copy_res(201): It was a dns record type: 1 class: 1
chilli[1271]: dns_copy_res(209): DNS: accounts.google.com
chilli[1271]: RADIUS queue-out id=0 idx=0
chilli[1271]: dns_copy_res(201): It was a dns record type: 1 class: 1
chilli[1271]: dns_copy_res(209): DNS: accounts.google.com
chilli[1271]: dns_copy_res(201): It was a dns record type: 1 class: 1
chilli[1271]: dhcp_receive_ip(3760): Address found
chilli[1271]: dhcp_receive_ip(3760): Address found
```

The reason of why the hotspot interface network is 192.168.10.0 is that I have reinstalled Kupiki with the default values of the script to see if it was any misconfiguration on my side. The only thing I've changed is the MySQL password, everything else is default.

Thank you very much for your help and your work.

pihomeserver, 2018-11-05:
I guess that's one of the reason now. The tun0 is up on 192.168.10.1 but the system is looking for 10.2.5.1
Can you please update portal configuration in the /var/share/nginx/portal/js/configuration file and set the new IP ? If you prefer to keep the previous one better is to reinstall all because i don't know what sides effects could be with 2 install at the same time

rubenmdh, 2018-11-05:
Hello, the current installation has overwritten the old one and the IP had already been overwritten.

The file looks like this (without having edited anything myself):
```
{
  "options": {
    "terms": {
      "active": true,
      "message": "I have read and I agree with the Terms and Conditions",
      "link": "Terms",
      "error": "You need to read and accept the Terms and Conditions before proceeding",
      "terms": "Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor incididunt ut labore et dolore magna aliqua. Ut enim ad minim veniam, quis nostrud exercitation ullamco laboris nisi ut aliquip ex ea commodo$
    },
    "userAuthentication": {
      "active": true,
      "type": "user"
    }
  },
  "chilliController": {
    "host": "192.168.10.1",
    "port": "3990",
    "interval": 60
  }
}
```

I can flash a clean Raspbian Lite image and reinstall from scratch if you are more comfortable with that.

pihomeserver, 2018-11-05:
So that means that on the first SD card, the Coova configuration is still on the previous IP. That sounds normal as i use sed instruction to search and replace IP. The old value is not found so the update is not performed.

For the limit, you can use freeradius attributes to define quota, bandwidth limit etc. Check freeradius documentation else please open a new ticket to not mix requests

closed 2018-11-06 by pihomeserver