pihomeserver/Kupiki-Hotspot-Script #145: Unknown name "Access-Period"

macalimosa, 2018-11-20:
#### Logs

```
(11) sql: ERROR: Failed to create the pair: Unknown name "Access-Period"
(11) sql: ERROR: Error parsing user data from database result
(11) sql: ERROR: Error retrieving check pairs for group client
```

### Your environment

- Hardware : Rasperberry pi 3 b+
- Operating system: Raspbian Sretch Full

### Additional comments

Would like to add attritube from daloradius profile but i got this errors when user trying to login.

macalimosa, 2018-11-20:
Solved by creating sqlcounter accessperiod

closed 2018-11-20 by macalimosa

pihomeserver, 2018-11-20:
Hello

Can you please share the updates and where you made them ?

macalimosa, 2018-11-20:
sure i add this code from /etc/freeradius/3.0/mods-available