pihomeserver/Kupiki-Hotspot-Script #146: acctstarttime wrong datime

macalimosa, 2018-11-21:
hello there, would like to ask why acctstarttime field from radacct table was not the same from my current datetime? i got - 30 minutes away from my current time. 

i did some research and they said "NAS and Freeradius datetime" must the same but i don't know where to find those configuration .

pihomeserver, 2018-11-21:
Really strange. I will investiguate with mine. Can you share your attributes configuration and parameters ?

macalimosa, 2018-11-22:
ok, i'd  just enabled "expire_on_login" from /etc/freeradius/3.0/sites-enabled authorize to be able to use the module, seems  works fine as well the query, but i had problem in acctstarttime.

macalimosa, 2018-11-22:
[image]

pihomeserver, 2018-11-22:
Can you runu both freeradius and coova in debug mode and compare timestamps from system, coova and freeradius ?
Also the gap between event timestamp and detail one seems ok as one is PST and the second not. Sounds as a problem of timezone and maybe then LOCAL configuration

macalimosa, 2018-11-26:
I've noticed datetime used the last time i used the pi e.g 3 days ago,also i resolved the issue when restarting freeradius and pi. so i need todo that every datetime incorrect :(

pihomeserver, 2018-11-26:
Did you check which date time is correct ? Don't you have an issue with the NTP sync or your local definition ?

macalimosa, 2018-11-27:
sorry i don't how to check chili time, i tried to restart chili and freeradius service but i got same result. i'm only resolved this issue is by rebooting my raspi, but when i did shutdown and reopen raspi still hotspot use the old datetime, even my local,mysql and Universal datetime is correct.

i have this timedatectl, all of this is correct
```
Local time: Tue 2018-11-27 22:23:15 PST
Universal time: Tue 2018-11-27 14:23:15 UTC
RTC time: n/a
Time zone: Asia/Manila (PST, +0800)
Network time on: yes
NTP synchronized: yes
RTC in local TZ: no
```

macalimosa, 2018-11-28:
Here what i'm thinking but i don't know how to do this
- run ntp sync before freeradius and chovachilli init
i read this from here https://www.cisco.com/c/en/us/td/docs/ios-xml/ios/sec_usr_radatt/configuration/15-s/sec-usr-radatt-15-s-book/sec-rad-att-55-event-timestamp.pdf

```
Before the Event-Timestamp attribute can be sent in accounting and authentication request packets, you must
configure the clock on the network device. For information about setting the clock on your network device,
see the “Performing BasicSystem Management”section in the “BasicSystem Management” chapter of Network
Management Configuration Guide.
To avoid configuring the clock on the network device every time the network device is reloaded, you can
enable the clock calendar-valid command. For information about this command, see the “Setting Time and
```

macalimosa, 2018-11-28:
after experimental solutions i solved the issue by running this codes
```
service chilli stop
service hostapd restart
service hostapd restart
service chilli start
```
now acctstarttime field in my database inserted  correctly
my  next is problem how to fix this in init.d

pihomeserver, 2018-11-28:
That's amazing that we did not found the issue before. As soon as i have my laptop back i will check on my system. A stop start of hostapd will kick users from the hotspot so that's not acceptable

closed 2019-05-18 by macalimosa