pihomeserver/Kupiki-Hotspot-Script #152: samsung phone doesn't get notification or auto launch captive portal:

sahtelnet117, 2018-12-11:
### Actual behavior

{Please write here}

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : {Please write here}
- Operating system: {Please write here}

### Additional comments

{Please write here, if there is something more to tell}

zdoucha, 2018-12-11:
Hello, I have an issue detected only on samsung s4 and j7 but ok on others (ios, lenovo, asus, huawei)
I installed a webserver with captive portal on a raspberry with a hotspot (nginx/apache2/lighttpd, hostapd, dnsmasq) when i connect to this wifi network a popup "Sign-in to network" is displayed or an auto launch of the captive portal page is done.
This behavior failed only on some samsung smartphone like s4 and j7, when i connect nothing happened!!! no popup nor auto display CP page
any proposition?

pihomeserver, 2018-12-12:
Duplicated with #153

closed 2018-12-12 by pihomeserver