pihomeserver/Kupiki-Hotspot-Script #155: [Question] Facebook Like/Check-in in lieu of free wifi?

Elite, 2018-12-14:
Can this be used to ask user to check-in/like my facebook page before using the free Wifi?

pihomeserver, 2018-12-14:
Not a function implemented  for now

Elite, 2018-12-16:
But this can be achieved by serving my facebook app page as splash screen? right?

pihomeserver, 2018-12-16:
Maybe. Try to replace HS_UAMFORMAT and HS_UAMHOMEPAGE by the facebook page

reigelgallarde, 2018-12-19:
I have built 4 types of hotspot solution, and this is the current one:   
https://mavispisowifisolutions.com/piso-wifi/  
And here's a video of what I have so far for this one: https://www.facebook.com/MavisPisoWifiSolutions/videos/353494265202403/
It's not finish yet but I'm updating it everyday I can.

With that said, I can honestly say that with just freeradius and coovachilli, you can't. Or to say you have a little chance.

However, you might be able to get it if you can make a script in relation to your goal.
A rough example is when you have a success hit of the "Like" button, create a username and password then automatically logged it in. For that you will have to write your own code for Kupiki-Hotspot-Script don't have it. 

If you're an advanced programmer, I would suggest to create your own radius as it would be easy to not need to create a username and password and just tell coova chilli to authenticate this device.. At least that's what I did, I'm not using freeradius but I have my own.

It's kind of interesting to implement it, but currently I don't have time.

Elite, 2018-12-20:
@reigelgallarde Do you have something for my need, free wifi access for a like/check-in to my facebook app. I am ready to pay for it.

12e1121, 2018-12-26:
Idea of Social Login (Facebook Login) in this case:
1. Hit "FB Login" button.
2. Call `FB.api` to retrieve user's details
3. Create freeradius's User Account using "`fb-$fbid"` , password `"$fbid"`
4. Login using `chilliController.logon("fb-$fbid", "$fbid")`

Problem Statement:
1. Required FB appId (which is quite easy to create and obtain)
2. HTTPS (SSL) is required to retrieve information using FB API
3. Required quite a lot of code changes in front/back end.

pihomeserver, 2018-12-26:
@12e1121 nice suggestion. The issue i see is that login/password are in the call. Maybe a requirement for social login could be to activate HTTPS support
I agree with the code change. I feel it as huge to have a social login option.
That seems to require a kind of middleware to first check the social login and then log into coova. I would like to avoid the creation of credentials within freeradius. Or maybe something linked to the MAC adress ?

reigelgallarde, 2018-12-29:
This issue is not related to the script actually, it's more on the implementation of the portal.
Should be addressed here: [Kupiki-Hotspot-Portal](https://github.com/Kupiki/Kupiki-Hotspot-Portal)

reigelgallarde, 2019-02-06:
Here's what I've done with coova chilli...

https://www.facebook.com/MavisPisoWifiSolutions/videos/2284367325134710/

I'm now working on how I can implement what op wants..

pihomeserver, 2019-02-06:
If it could be shared that would help all

reigelgallarde, 2019-02-07:
I will share what the OP wants after I'm finish...

reigelgallarde, 2019-02-12:
@pihomeserver, I have a problem and thought you could help me haha.

I'm working on this but got stuck.
I've seen you around on node-radius, so I think you might have an idea.

after validating the "like" event on facebook. I will send a Access-Request...
then on my Access-Accept, I want to use "Session-Timeout" in attributes but currently have no success.

pihomeserver, 2019-02-12:
My first purpose was about register. Now it works but without node-radius. For Social Network validation i wanted to use sign on on from external sites. 
About your like on facebook, do you know how to check if someone made a like ? If yes then build a node middleware that will receive Coova requests, do stuff with it (check facebook ?) and send back an Access-Accept ou Access-Reject.
I was able to receive request from coova, do some controls but the reply from the middleware to Coova was always misunderstood in Coova. I stop fast and use another way to work on the register feature. I will come back later on that
If you want to use a middleware in coova, build it, make i run in local (port 5000 in my example) and add next lines in /etc/chilli/config 
```
HS_AAA=http
HS_UAMAAAURL=http://127.0.0.1:5000/
```
You should then receive all requests from Coova. In the basic test i made, i pushed the request to freeradius and tried to push back the answer from FR to Coova.

reigelgallarde, 2019-02-13:
Oh interesting... I've really wanted to build an AAA for coova... but don't know where to start...
So what I did, is I've built my own radius server...

pihomeserver, 2019-02-13:
If you want to work on that again we can discuss and share experience in a dedicated ticket