pihomeserver/Kupiki-Hotspot-Script #157: How can I get the original Coovachilli captive portal?

desevensa, 2019-01-31:
### Actual behavior

Runs Ok with new Captive portal for wifi clients

### Expected behavior

Install the old-fashioned Captive portal

### Additional comments

I want make modifications for that code in HTML

pihomeserver, 2019-02-01:
Not sure to understand. You want the default portal from Coova ?

desevensa, 2019-02-01:
Yes, exactly that.

pihomeserver, 2019-02-01:
Be sure that Haserl is installed and then update/replace following parameters in config file : 
```
HS_UAMSERVER=$HS_UAMLISTEN
HS_UAMUIPORT=4990
HS_UAMFORMAT=http://\$HS_UAMLISTEN:\$HS_UAMUIPORT/www/login.chi
HS_UAMHOMEPAGE=http://\$HS_UAMLISTEN:\$HS_UAMPORT/www/coova.html
```

desevensa, 2019-02-01:
Thanks a lot. I will do it. :+1:

desevensa, 2019-02-03:
@pihomeserver Thank you. I get it now the default coova portal.

closed 2019-02-03 by desevensa