pihomeserver/Kupiki-Hotspot-Script #158: Problem Installing (missing mariadb)

raspblu, 2019-02-01:
I am also having an error that was mentioned awhile back:

```
::: Starting MySql service
Failed to restart mysql.service: Unit mysql.service not found.
::: 
::: Something went wrong with the last command. Please check the log file
::: 
```
I am using raspberry pi 3b+ on Stretch (non-lite version). Is it possible to successfully install this on a non-lite version of raspbian? I need the desktop for other processing. Also, as a side question, is it possible direct the captive portal to an offline php website on the raspberry pi? If so, how would I go about it? thank you.

pihomeserver, 2019-02-01:
Did you investigate ? Is the service available ? After a reboot do you services MySQL, Freeradius and Coova running ?

raspblu, 2019-02-01:
No, it won't even install at all fully:

```
The following packages have unmet dependencies:
 mariadb-server : Depends: mariadb-server-10.1 (>= 10.1.37-0+deb9u1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
That's the problem I am stuck with now. I am on Raspbian Stretch Debian 9.6, fresh install...and for some reason I cannot get mariadb-server installed?

pihomeserver, 2019-02-01:
To install `mariadb-server` package, it depends on the package `mariadb-server-10.1` [(check here)](https://packages.debian.org/stretch/mariadb-server)
No idea why on the full Raspbian the system is not able to install dependencies automatically. Try manually. Else use the Lite release.

pihomeserver, 2019-02-01:
For the PHP site you can create your own and define it as default. It seems that the issue will be that Coova requires internet (https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/75#issuecomment-426309851)

raspblu, 2019-02-01:
I got it working now by installing fresh Raspbian Stretch OS. Can you please further clarify how I would be able to redirect to an index.php after user login to hotspot landing page? I am a total noob. Where is the default to be defined located at?

pihomeserver, 2019-02-01:
just replace the website located within /usr/nginx/share/portal by whatever you want. If your site is located somewhere else update the configuration in /etc/nginx/sites-enabled/portal

closed 2019-02-13 by pihomeserver