pihomeserver/Kupiki-Hotspot-Script #160: Impossible de créer un compte utilisateur, croix rouge dans Daloradius

indian65, 2019-02-04:
### Actual behavior

{Please write here}

### Expected behavior
Impossible de créer un user, croix rouge systématiquement, quelque soit le type d'identification, idem avec un user créer en ligne de commande
Les services freeradius fonctionnent. Là, je sèche.
{Please write here}

#### Configuration parameters (if changed in the script)
Script installé sans modification
{Please write here}

#### Logs
info: Dropping packet without response because of error. Receveid Accounting request packet ffrom client 127.0.0.1 with invalid request authenticator ( shared secret is incorrect)
```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : Raspberry pi 3 b+
- Operating system: Raspbian stretch

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2019-02-04:
Hello
First be sure that the value of HS_RADSECRET in /etc/chilli/config is the same than secret in /etc/freeradius/3.0/clients.conf (2 values should be uncomment)
Then stop freeradius service, start freeradius in debug mode and run the radtest command to check that all is running fine (look for an example here https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/10#issuecomment-262168674)

indian65, 2019-02-04:
The two values are differents, i change for the same key, (the key of freeradius) but when i test a new user I have that
pi@kupikihotspot:~ $ echo "insert into radcheck (username, attribute, op, value) values ('usertest', 'Cleartext-Password', ':=', 'passwd');" | mysql -u root -praspbian radius
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)

pihomeserver, 2019-02-04:
Password for the db is wrong. Check the value in the script and replace raspbian with the correct one. Should be pihotspot by default

indian65, 2019-02-05:
The same
pi@kupikihotspot:~ $ echo "insert into radcheck (username, attribute, op, value) values ('usertest', 'Cleartext-Password', ':=', 'passwd');" | mysql -u root -pihotspot radius
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)

pihomeserver, 2019-02-05:
You forget the p before the password ! It’s -ppihotspot

pihomeserver, 2019-02-05:
That’s normal you use testing123 as the secret key instead of the real one.

indian65, 2019-02-05:
It's work, 
When i create a new user by step, it's no recognize but when i use quick add user option, it's work.
Thank's for your quickly responses.
Merci beaucoup pour votre réactivité.

pihomeserver, 2019-02-05:
I guess there is something wrong in the way you try to create the user. Often the problem is in the attribute creation to add a check for the password

pihomeserver, 2019-02-05:
The installation is not related to the way the users are created. All is handle by Daloradius. Once the user is created, be sure that in the radius database you have something like bellow
[image]

indian65, 2019-02-05:
1/I tried with my antenna, the driver is recognized and the antenna activated (LED lit) but no visible wifi network.
2/ for the radius database, is it possible to give me the path of the file, I am a noob.

pihomeserver, 2019-02-05:
If you are not with little linux experience it will be difficult to achieve your goal
For the antena did you updated the script with the new interface ?
About the database it’s not a file but a database in the mariadb instance. Look at google about a GUI tool for your OS to connect to your instance (after allowing remote root access)

indian65, 2019-02-05:
For the antenna, I changed the two values
LAN_INTERFACE: wlan1
LAN_WIFI_DRIVER:ath9k_htc (atheros 9271)
Maybe I'm forgetting something

pihomeserver, 2019-02-05:
If your antenna is up and running on wlan1 no it should be ok. Are you sure that the wifi key could work as an access point ?

indian65, 2019-02-05:
`iw list
Wiphy phy1
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x1 RX 0x1
	Configured Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
		 * outside context of a BSS
	Band 1:
		Capabilities: 0x116e
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
[... 160 lines cut ...]
		 * start_p2p_device
		 * connect
		 * disconnect
		 * crit_protocol_start
		 * crit_protocol_stop
		 * Unknown command (122)
	Supported TX frame types:
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * managed: 0x40 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
	valid interface combinations:
		 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
		   total <= 3, #channels <= 2
		 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
		   total <= 4, #channels <= 1
	Device supports scan flush.
pi@raspberrypi:~ $ 
`

pihomeserver, 2019-02-05:
Seems to support AP. To check on the web if hostapd can handle the hardware as it’s it who expose the SSID

indian65, 2019-02-06:
J'ai trouvé, l'antenne fonctionne avec le driver originel, j'ai reconfigurer le fichier hostapd.conf, reboot et mon reseau est visible et fonctionnelle.
la clé a été cette page
http://benoit.ofoofo.net/2012/07/21/hostapd-configuration-for-a-802-11n-ap-with-ath9k/

indian65, 2019-02-08:
A new issue when i want to create a new user
FreeRADIUS Version 3.0.12
Copyright (C) 1999-2016 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius/dictionary
including dictionary file /usr/share/freeradius/dictionary.dhcp
including dictionary file /usr/share/freeradius/dictionary.vqp
including configuration file /etc/freeradius/3.0/radiusd.conf
Unable to open file "/etc/freeradius/3.0/radiusd.conf": Permission denied
Errors reading or parsing /etc/freeradius/3.0/radiusd.conf

pihomeserver, 2019-02-08:
Look at the error message, i guess that you are not running the command as root. It should be better to learn about user creation in daloradius to avoid the use of root

indian65, 2019-02-08:
I test in root
no problème with freeradius, when i test a new user with daloradius, acces-accept, but when i tried to connect with the new user, i have password or login incorrect

indian65, 2019-02-08:
I create a new user, and test
Executed:
echo User-Name='condorinette',User-Password='condorinette' | radclient -c '1' -n '3' -r '3' -t '3' -x '127.0.0.1:1812' 'auth' 'PrUHmdaHkHFDCxmT85RPWlKoHyuQWOXm' 2>&1

Results:
Sent Access-Request Id 235 from 0.0.0.0:46805 to 127.0.0.1:1812 length 52
User-Name = "condorinette"
User-Password = "condorinette"
Cleartext-Password = "condorinette"
Received Access-Accept Id 235 from 127.0.0.1:1812 to 0.0.0.0:0 length 20

but when i want to connect with this user

pihomeserver, 2019-02-08:
Stop freeradius service and start freeradius in debug mode. You will see the error message. Pretty sure the secret key is the issue

indian65, 2019-02-08:
The secret key are the same in the two folders
When i start freeradius in debug mode
`Station-Id}', '%{Calling-Station-Id}', '', '%{Service-Type}', '%{Framed-Protocol}', '%{Framed-IP-Address}')"
     }
     interim-update {
     	query = "UPDATE radacct SET acctupdatetime  = (@acctupdatetime_old:=acctupdatetime), acctupdatetime  = FROM_UNIXTIME(%{integer:Event-Timestamp}), acctinterval    = %{integer:Event-Timestamp} - UNIX_TIMESTAMP(@acctupdatetime_old), framedipaddress = '%{Framed-IP-Address}', acctsessiontime = %{%{Acct-Session-Time}:-NULL}, acctinputoctets = '%{%{Acct-Input-Gigawords}:-0}' << 32 | '%{%{Acct-Input-Octets}:-0}', acctoutputoctets = '%{%{Acct-Output-Gigawords}:-0}' << 32 | '%{%{Acct-Output-Octets}:-0}' WHERE AcctUniqueId = '%{Acct-Unique-Session-Id}'"
     }
     stop {
     	query = "UPDATE radacct SET acctstoptime	= FROM_UNIXTIME(%{integer:Event-Timestamp}), acctsessiontime	= %{%{Acct-Session-Time}:-NULL}, acctinputoctets	= '%{%{Acct-Input-Gigawords}:-0}' << 32 | '%{%{Acct-Input-Octets}:-0}', acctoutputoctets = '%{%{Acct-Output-Gigawords}:-0}' << 32 | '%{%{Acct-Output-Octets}:-0}', acctterminatecause = '%{Acct-Terminate-Cause}', connectinfo_stop = '%{Connect-Info}' WHERE AcctUniqueId = '%{Acct-Unique-Session-Id}'"
     }
    }
   }
   post-auth {
   	reference = ".query"
   	query = "INSERT INTO radpostauth (username, pass, reply, authdate) VALUES ( '%{SQL-User-Name}', '%{%{User-Password}:-%{Chap-Password}}', '%{reply:Packet-Type}', '%S')"
   }
  }
rlm_sql (sql): Driver rlm_sql_mysql (module rlm_sql_mysql) loaded and linked
Creating attribute SQL-Group
  # Loaded module rlm_logintime
  # Loading module "logintime" from file /etc/freeradius/3.0/mods-enabled/logintime
  logintime {
  	minimum_timeout = 60
  }
  # Loaded module rlm_exec
  # Loading module "exec" from file /etc/freeradius/3.0/mods-enabled/exec
  exec {
  	wait = no
  	input_pairs = "request"
  	shell_escape = yes
  	timeout = 10
  }
  # Loaded module rlm_files
  # Loading module "files" from file /etc/freeradius/3.0/mods-enabled/files
  files {
  	filename = "/etc/freeradius/3.0/mods-config/files/authorize"
  	acctusersfile = "/etc/freeradius/3.0/mods-config/files/accounting"
  	preproxy_usersfile = "/etc/freeradius/3.0/mods-config/files/pre-proxy"
  }
  # Loaded module rlm_linelog
  # Loading module "linelog" from file /etc/freeradius/3.0/mods-enabled/linelog
  linelog {
  	filename = "/var/log/freeradius/linelog"
  	escape_filenames = no
  	syslog_severity = "info"
  	permissions = 384
  	format = "This is a log message for %{User-Name}"
  	reference = "messages.%{%{reply:Packet-Type}:-default}"
  }
  # Loading module "log_accounting" from file /etc/freeradius/3.0/mods-enabled/linelog
  linelog log_accounting {
  	filename = "/var/log/freeradius/linelog-accounting"
  	escape_filenames = no
  	syslog_severity = "info"
  	permissions = 384
  	format = ""
  	reference = "Accounting-Request.%{%{Acct-Status-Type}:-unknown}"
  }
  # Loaded module rlm_chap
  # Loading module "chap" from file /etc/freeradius/3.0/mods-enabled/chap
  # Loaded module rlm_pap
  # Loading module "pap" from file /etc/freeradius/3.0/mods-enabled/pap
  pap {
  	normalise = yes
  }
  # Loaded module rlm_realm
  # Loading module "IPASS" from file /etc/freeradius/3.0/mods-enabled/realm
  realm IPASS {
  	format = "prefix"
  	delimiter = "/"
[... 708 lines cut ...]
  	type = "acct"
  	ipv6addr = ::
  	port = 0
   limit {
   	max_connections = 16
   	lifetime = 0
   	idle_timeout = 30
   }
}
listen {
  	type = "auth"
  	ipaddr = 127.0.0.1
  	port = 18120
}
Listening on coa address * port 1700 bound to server coa
Listening on auth address * port 1812 bound to server default
Listening on acct address * port 1813 bound to server default
Listening on auth address :: port 1812 bound to server default
Listening on acct address :: port 1813 bound to server default
Listening on auth address 127.0.0.1 port 18120 bound to server inner-tunnel
Listening on proxy address * port 37393
Listening on proxy address :: port 33535
Ready to process requests

`

indian65, 2019-02-09:
I don't know why, but my new user is accept this morning.
Thanks for your help.
Very good job.

closed 2019-02-09 by indian65