pihomeserver/Kupiki-Hotspot-Script #164: Command "python setup.py egg_info" failed with error code 1

arieswebdev001, 2019-04-03:
### Actual behavior

Error occurred when running this sudo chmod +x pihotspot.sh && sudo ./pihotspot.sh
it says: Command "python setup.py egg_info" failed with error code 1

### Expected behavior

It breaks the installation

#### Configuration parameters (if changed in the script)

Nothing was changed

#### Logs

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-w0R90I/pynacl/
:::
::: Something went wrong with the last command. Please check the log file
:::

### Your environment

- Hardware : Raspberry PI 3 b+
- Operating system: raspbian stretch lite

### Additional comments

pihomeserver, 2019-04-03:
Hello
Can you please provide last lines of the file /var/log/pihotspot.log ?

12e1121, 2019-04-05:
@pihomeserver I have the same problem too. The logs show:

```
::: Install Docker
::: Create dedicated user kupiki
::: Add user kupiki to docker group
::: Install docker-compose
:::
::: Something went wrong with the last command. Please check the log file
:::
```

The exact error should be in this line
```
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

and scroll back the error, found out that might be less dependencies when installing **docker-compose**:

```
No package 'libffi' found
    c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
                     ^
    compilation terminated.

```

After some googling, found out that [this](https://medium.com/@melzoghbi/deploy-iot-edge-on-raspberry-pi-e9cc9004ea3a) step 3, installing the dependencies solve the problem.

I ran:

`sudo apt-get install build-essential libssl-dev libffi-dev python3-dev`

then:

`sudo pip install docker-compose`

And finally successfully installed. 

```
Successfully installed PyYAML-3.13 backports.ssl-match-hostname-3.7.0.1 bcrypt-3.1.6 cached-property-1.5.1 cffi-1.12.2 docker-3.7.2 docker-compose-1.24.0 docker-pycreds-0.4.0 dockerpty-0.4.1 docopt-0.6.2 functools32-3.2.3.post2 idna-2.7 jsonschema-2.6.0 paramiko-2.4.2 pycparser-2.19 pynacl-1.3.0 requests-2.20.1 texttable-0.9.1 websocket-client-0.56.0
```

Not sure will this help?

KahShion, 2019-04-05:
My best guest is we are lessing this package:

`libffi-dev`

pihomeserver, 2019-04-05:
@KahShion thank you that's really helpful. I released a new version of the script!

RezzZ, 2019-06-12:
I got the same problem installing docker_compose.
I installed libffi-dev:
```
sudo apt-get install libffi-dev
```
and now the install fails with
Running setup.py bdist_wheel for cryptography ... error
```
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```
and finally
```
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8RR4Ag/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-1shSuo-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-8RR4Ag/cryptography/
```

pihomeserver, 2019-06-23:
@RezzZ pretty strange as this package is already included in the installation (line 335 of the script)

closed 2019-06-23 by pihomeserver

RezzZ, 2019-06-24:
@pihomeserver sorry, my problems although similar didn't have the same context. I was just trying to install docker-compose, not using the Kupiki-Hotspot-Script. this issue did help me solve the problem though so thanks for that ;)

dwg-andy, 2019-07-11:
> I got the same problem installing docker_compose.
> I installed libffi-dev:
> 
> ```
> sudo apt-get install libffi-dev
> ```
> 
> and now the install fails with
> Running setup.py bdist_wheel for cryptography ... error
> 
> ```
> error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
> ```
> 
> and finally
> 
> ```
>     error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
> 
>     ----------------------------------------
>   Can't rollback cryptography, nothing uninstalled.
> Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8RR4Ag/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-1shSuo-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-8RR4Ag/cryptography/
> ```
> 
> edit:
> after installing libssl-dev as well install completed without problems:
> 
> ```
> sudo apt-get install libssl-dev libffi-dev
> ```

Installing on Chromebook aarch64. this fixed it for me