pihomeserver/Kupiki-Hotspot-Script #165: Update Kupiki error script

nextgen1979, 2019-04-11:
Hello , i have error when i try to update :

pi@ZeeDijk126-Hotspot:~ $ sudo /etc/kupiki/kupiki_updater.sh
:::
::: You are root - OK
::: Auto updating of the script 
:::
::: You are root - OK
::: Running updater
::: Preparing log file
::: Log file already exists. Creating a backup.
::: 
::: Creating the log file
::: 
::: Log file created : /var/log/kupiki_updater.log
::: Use command 'tail -f /var/log/kupiki_updater.log' in a new console to get updater process details
::: Getting current installed version
::: Getting latest version of the updater script
::: Latest version : 2.1.1
::: Current installed version : 2.0.13
::: 
::: Upgrading to version 2.1.0
::: 
::: Cloning Pi Hotspot portal backend project
::: Correct configuration for Collectd daemon
docker is /usr/bin/docker
::: Install docker-compose
Collecting docker-compose
  Using cached https://files.pythonhosted.org/packages/51/56/5745e66b33846e92a8814466c163f165a26fadad8b33afe381e8b6c3f652/docker_compose-1.24.0-py2.py3-none-any.whl
Collecting docker[ssh]<4.0,>=3.7.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/48/68/c3afca1a5aa8d2997ec3b8ee822a4d752cf85907b321f07ea86888545152/docker-3.7.2-py2.py3-none-any.whl
Collecting backports.ssl-match-hostname>=3.5; python_version < "3.5" (from docker-compose)
  Using cached https://www.piwheels.org/simple/backports-ssl-match-hostname/backports.ssl_match_hostname-3.7.0.1-py2.py3-none-any.whl
Collecting websocket-client<1.0,>=0.32.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/29/19/44753eab1fdb50770ac69605527e8859468f3c0fd7dc5a76dd9c4dbd7906/websocket_client-0.56.0-py2.py3-none-any.whl
Requirement already satisfied: ipaddress>=1.0.16; python_version < "3.3" in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting PyYAML<4.3,>=3.10 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/9e/a3/1d13970c3f36777c583f136c136f804d70f500168edc1edea6daa7200769/PyYAML-3.13.tar.gz
Collecting dockerpty<0.5,>=0.4.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/8d/ee/e9ecce4c32204a6738e0a5d5883d3413794d7498fe8b06f44becc028d3ba/dockerpty-0.4.1.tar.gz
Collecting texttable<0.10,>=0.9.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/02/e1/2565e6b842de7945af0555167d33acfc8a615584ef7abd30d1eae00a4d80/texttable-0.9.1.tar.gz
Requirement already satisfied: enum34<2,>=1.0.4; python_version < "3.4" in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting requests!=2.11.0,!=2.12.2,!=2.18.0,<2.21,>=2.6.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/ff/17/5cbb026005115301a8fb2f9b0e3e8d32313142fe8b617070e7baad20554f/requests-2.20.1-py2.py3-none-any.whl
Requirement already satisfied: six<2,>=1.3.0 in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting jsonschema<3,>=2.5.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/77/de/47e35a97b2b05c2fadbec67d44cfcdcd09b8086951b331d82de90d2912da/jsonschema-2.6.0-py2.py3-none-any.whl
Collecting docopt<0.7,>=0.6.1 (from docker-compose)
  Downloading https://www.piwheels.org/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl
Collecting cached-property<2,>=1.2.0 (from docker-compose)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
::: 
::: Something went wrong with the last command. Please check the log file
::: 
pi@ZeeDijk126-Hotspot:~ $ 

Do you have an idea ?

Thanks in advance

nextgen1979, 2019-04-11:
ok when i try other sudo -i . and sudo command i have this :  

pi@ZeeDijk126-Hotspot:~ $ sudo -i
root@ZeeDijk126-Hotspot:~# cd etc
-bash: cd: etc: No such file or directory
root@ZeeDijk126-Hotspot:~# cd /etc/
root@ZeeDijk126-Hotspot:/etc# cd kupiki/
root@ZeeDijk126-Hotspot:/etc/kupiki# ./kupiki_updater.sh 
:::
::: You are root - OK
::: Auto updating of the script 
:::
::: You are root - OK
::: Running updater
::: Preparing log file
::: Log file already exists. Creating a backup.
::: 
::: Creating the log file
::: 
::: Log file created : /var/log/kupiki_updater.log
::: Use command 'tail -f /var/log/kupiki_updater.log' in a new console to get updater process details
::: Getting current installed version
::: Getting latest version of the updater script
::: Latest version : 2.1.1
::: Current installed version : 2.0.13
::: 
::: Upgrading to version 2.1.0
::: 
::: Cloning Pi Hotspot portal backend project
::: Correct configuration for Collectd daemon
docker is /usr/bin/docker
::: Install docker-compose
Collecting docker-compose
  Using cached https://files.pythonhosted.org/packages/51/56/5745e66b33846e92a8814466c163f165a26fadad8b33afe381e8b6c3f652/docker_compose-1.24.0-py2.py3-none-any.whl
Collecting docker[ssh]<4.0,>=3.7.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/48/68/c3afca1a5aa8d2997ec3b8ee822a4d752cf85907b321f07ea86888545152/docker-3.7.2-py2.py3-none-any.whl
Collecting backports.ssl-match-hostname>=3.5; python_version < "3.5" (from docker-compose)
  Using cached https://www.piwheels.org/simple/backports-ssl-match-hostname/backports.ssl_match_hostname-3.7.0.1-py2.py3-none-any.whl
Collecting websocket-client<1.0,>=0.32.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/29/19/44753eab1fdb50770ac69605527e8859468f3c0fd7dc5a76dd9c4dbd7906/websocket_client-0.56.0-py2.py3-none-any.whl
Requirement already satisfied: ipaddress>=1.0.16; python_version < "3.3" in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting PyYAML<4.3,>=3.10 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/9e/a3/1d13970c3f36777c583f136c136f804d70f500168edc1edea6daa7200769/PyYAML-3.13.tar.gz
Collecting dockerpty<0.5,>=0.4.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/8d/ee/e9ecce4c32204a6738e0a5d5883d3413794d7498fe8b06f44becc028d3ba/dockerpty-0.4.1.tar.gz
Collecting texttable<0.10,>=0.9.0 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/02/e1/2565e6b842de7945af0555167d33acfc8a615584ef7abd30d1eae00a4d80/texttable-0.9.1.tar.gz
Requirement already satisfied: enum34<2,>=1.0.4; python_version < "3.4" in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting requests!=2.11.0,!=2.12.2,!=2.18.0,<2.21,>=2.6.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/ff/17/5cbb026005115301a8fb2f9b0e3e8d32313142fe8b617070e7baad20554f/requests-2.20.1-py2.py3-none-any.whl
Requirement already satisfied: six<2,>=1.3.0 in /usr/lib/python2.7/dist-packages (from docker-compose)
Collecting jsonschema<3,>=2.5.1 (from docker-compose)
  Using cached https://files.pythonhosted.org/packages/77/de/47e35a97b2b05c2fadbec67d44cfcdcd09b8086951b331d82de90d2912da/jsonschema-2.6.0-py2.py3-none-any.whl
Collecting docopt<0.7,>=0.6.1 (from docker-compose)
  Using cached https://www.piwheels.org/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl
Collecting cached-property<2,>=1.2.0 (from docker-compose)
  Downloading https://files.pythonhosted.org/packages/3b/86/85c1be2e8db9e13ef9a350aecd6dea292bd612fa288c2f40d035bb750ded/cached_property-1.5.1-py2.py3-none-any.whl
Collecting docker-pycreds>=0.4.0 (from docker[ssh]<4.0,>=3.7.0->docker-compose)
  Downloading https://files.pythonhosted.org/packages/f5/e8/f6bd1eee09314e7e6dee49cbe2c5e22314ccdb38db16c9fc72d2fa80d054/docker_pycreds-0.4.0-py2.py3-none-any.whl
Collecting paramiko>=2.4.2; extra == "ssh" (from docker[ssh]<4.0,>=3.7.0->docker-compose)
  Downloading https://files.pythonhosted.org/packages/cf/ae/94e70d49044ccc234bfdba20114fa947d7ba6eb68a2e452d89b920e62227/paramiko-2.4.2-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 857kB/s 
Collecting urllib3<1.25,>=1.21.1 (from requests!=2.11.0,!=2.12.2,!=2.18.0,<2.21,>=2.6.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/62/00/ee1d7de624db8ba7090d1226aebefab96a2c71cd5cfa7629d6ad3f61b79e/urllib3-1.24.1-py2.py3-none-any.whl (118kB)
    100% |████████████████████████████████| 122kB 1.4MB/s 
Collecting idna<2.8,>=2.5 (from requests!=2.11.0,!=2.12.2,!=2.18.0,<2.21,>=2.6.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/4b/2a/0276479a4b3caeb8a8c1af2f8e4355746a97fab05a372e4a2c6a6b876165/idna-2.7-py2.py3-none-any.whl (58kB)
    100% |████████████████████████████████| 61kB 1.6MB/s 
Collecting chardet<3.1.0,>=3.0.2 (from requests!=2.11.0,!=2.12.2,!=2.18.0,<2.21,>=2.6.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl (133kB)
[... 51 lines cut ...]
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 699, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 725, in install_item
[... cut ...]
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-OB9BQC/pynacl/
::: 
::: Something went wrong with the last command. Please check the log file
::: 
root@ZeeDijk126-Hotspot:/etc/kupiki# 

but other error ...

What i'm doing wrong ?

Thanks in advance

nextgen1979, 2019-04-11:
And for now i don't see anymmore the hotspot wifi .  Thanks for your help

pihomeserver, 2019-04-18:
Ouch. Seems the updater script has bugs ... It's always complex to test in particular on a very slow Raspberry. Sorry for that.
From what i see, the problem is in the install process of Docker. Does the `docker` command exist (from command line) ?
We can check commands to be executed one by one

nextgen1979, 2019-04-18:
If you wish i can test for you , i have a lot of rpi 3 and 3b+   so i'm free to test all that .

closed 2019-06-26 by pihomeserver