pihomeserver/Kupiki-Hotspot-Script #166: Installation setup

nextgen1979, 2019-04-13:
I have a Pi3 i have download and install . raspbian strech lite from april 2019
make apt get update et upgrade
i have install git
i have download the script and edit the script , i have just modify for https portail ( Y ) 
run the script 
at the end with no error 

i try to connect to the portail but .  have blank page  .

i connect to daloradius from my lan but have a lot of error to the log access . and other setting . . 

I have done something wrong ?

i have try  all with the pi user . or with sudo-i 

Thanks in advance for your help 

Nicolas

pihomeserver, 2019-04-18:
@nextgen1979 I tried it this morning and also have the issue. It seems that the portal gateway is not reachable. May be an iptable rule missing to allow HTTPS traffic on port 443.

pihomeserver, 2019-04-19:
@nextgen1979 : can you please do following actions : 
- Stop Chilli with `service stop chilli`
- Update /etc/chilli/config by adding 443 to line `HS_TCP_PORTS="5000 80"`. Should be `HS_TCP_PORTS="5000 80 443"`
- Start Chilli with `service start chilli`

nextgen1979, 2019-04-19:
ok i have test

service chilli stop 
modify the config 
service chilli start 

now i have the portail correct

i will create user . and test connection 

i have see a lot of deprecated package . I will try to modify with new one  and if it's work i will send you the modification for the script 

Thanks

Nicolas

nextgen1979, 2019-04-19:
ok 
i don't have connection 

i try to enter with an user and password , click on accept terms and condition but after that i have a circle blink inside another circle and that it .

pihomeserver, 2019-04-19:
If you only display the portal and wait few seconds do you see an error message (or in tje developper console of the browser) ?

nextgen1979, 2019-04-21:
its not a browser page it's the portail of the wifi so i have wait a long time but nothing .  i have try from a mac . , i will try from a pc . 

i have start to write down the deprecated package in a list 

Nicolas
[image]