pihomeserver/Kupiki-Hotspot-Script #167: use outdoor AP instead of built in wlan0

reigelgallarde, 2019-04-18:
@pihomeserver I don't have a script for this now, so I can't do a PR.
I'm discussing it here anyway so you might be able to create the script instead.

**TLDR;**
This will enable raspberry pi to get a wider range wifi.

**Additional Hardware Requirements:**

- 1x USB to LAN ( be careful buying cheaper one, they have limited speed )
- 1x Router/Repeater that has AP mode or at least DHCP server can be disabled. (wavlink n300 is one good example). _For discussion purpose, lets call this just AP._

**Setup:**
You need to check the manual of your chosen AP hardware. Then make sure it is repeater mode or AP mode and DHCP server is disabled.
Now connect USB LAN to raspberry pi and connect your AP to it. 
[image]
Click on the image for a larger view.
This should now be detected as eth1 in raspberry pi.
There's a script from here https://github.com/oblique/create_ap/issues/61#issuecomment-71991145, this can be used to redirect internet from eth0 to eth1.
You need to try this first that eth1 has internet by connecting to your AP's SSID.
If there's internet, your pihotspot.sh should work by replacing wlan0 with eth1.
A couple enhancements though, you need to disable or remove hostapd, also disable wlan0.

**Note**
For test purpose, any repeater should do. If that repeater is not managing DHCP. I'm using a cheap repeater that costs around 8 usd for test purpose.

reigelgallarde, 2019-04-18:
I'm inviting too @droecoleman and @12e1121 in this discussion. This might interest you both.

droecoleman, 2019-04-18:
@reigelgallarde thanks man, i will keep you posted on the progress.

12e1121, 2019-04-18:
@reigelgallarde Interesting. I will keep you updated.

droecoleman, 2019-04-25:
@reigelgallarde can you elaborate how to do it? i can not make it work.

reigelgallarde, 2019-04-26:
@droecoleman you should describe more on what you have and what you have done...

droecoleman, 2019-04-26:
@reigelgallarde thanks for your response man. i appreciate it.