pihomeserver/Kupiki-Hotspot-Script #168: Script fails to run through

Tesseract83233, 2019-04-26:
### Actual behavior

Script stops with error, see attached screenshots please. 

[image]
[image]

### Expected behavior

Script should run through

#### Configuration parameters (if changed in the script)

default installation

### Your environment

- Hardware : Pi 3 B+, tried with multiple SD cards
- Operating system: tried both Raspbian Stretch Lite 2018-11-13 and 2019-04-09.

### Additional comments
Sorry, don't have a lot of experience. had it working before, but wanted to do a fresh install with the new self-register functionality. Can't access log file at the moment.

pihomeserver, 2019-04-26:
You have to be root to execute the script

Tesseract83233, 2019-04-26:
> 
> 
> You have to be root to execute the script
thanks for your answer. 
quite sure i was root every single time. tried it 4x... 
if you're sure, I'm happy to try again.

pihomeserver, 2019-04-26:
Maybe you were but then your screenshot shows that for this try you were not
@nextgen1979 : as you made a successful install yesterday, any attention point for the install ?

nextgen1979, 2019-04-26:
i will make a documentation to night . 

but for the install   only root  and  git install of course .

RI-Kai, 2019-04-26:
I have exactly the same problem. I executed the script as root.

pihomeserver, 2019-04-26:
I guess you missed something in the installation process. I have just restarted from scratch an installation and all went well : 
```
::: Congratulation ! You now have your hotspot ready !
:::
::: - Wifi Hotspot available : kupikihotspot
::: - For the user management, please connect to http://192.168.1.16/ or http://kupikihotspot.local/
:::   (login : administrator / password : radius)
```

So either you are not root or another option is that you used another distro that the Raspbian **Lite** release (i used the one from April)

RI-Kai, 2019-04-26:
I don't think I missed something, but of course I can't guarantee.
What I have done: I executed "pip install docker-compose" manually and then rerun the script. That worked now.
Thanks for this amazing project!

Tesseract83233, 2019-05-04:
just had time to try it again. it fails in the same way. I have root privileges, tried it on 2 different Pis and 3 different sd cards. 
something's going wrong and i don't know what...

Tesseract83233, 2019-05-04:
got it working. 
after a bit of research, if found this post about others having issues with the latest docker-compose install (1.24): https://github.com/docker/compose/issues/6617 
i edited the setup script to use 1.23 and it's working fine now. 
[image]
can't really explain why this isn't an issue for everyone, but it certianly was for me...

closed 2019-05-04 by Tesseract83233

pinoyoragon, 2019-05-17:
Still an issue to me, i followed Tesseract83233's advice to force 1.23 but fails on

`Collecting paramiko>=2.4.2; extra == "ssh" (from docker[ssh]<4.0,>=3.7.0->docker-compose~=1.23)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)`

mickrow, 2019-06-24:
not sure if this is related, but my script doesnt run through anymore either.
RPi 3 B+, stretch lite as of 2019-04-08 and also previous stretch, latest updates, via sudo and root, both clean SD and also with berryboot, also tried https NO and YES.

can't get passed the "Install Docker", which were the last words of the log file.
The last output from the script:
executing docker install script, commit: 2f4ae48
+sh -c apt-get update -qq >/dev/null
+sh -c apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
E: Sub-process /urs/bin/dpkg returned an error code (1)

it might be a similar issue mentioned above with a changed docker version meantime, however i have no clou how to tackle it, since that "pip install docker" -  line does not seem to be in this script nor in the get-docker.sh (which most likely will be overwritten anyways).

if you could point me into the right direction, would be great.

pihomeserver, 2019-06-26:
Ok then i guess i have to update the script to force the version even if that could be interesting to understand what kind of change occured to generate the error