pihomeserver/Kupiki-Hotspot-Script #169: hostapd fail to restart

indian65, 2019-05-14:
#### Configuration parameters (if changed in the script)

I just change the name of the hotspot and the wlan0 by wlan1
I tried to run the script, but everytime it's stop when it's restart hostapd

#### Logs of journalctl -xe

root@raspberrypi:~# journalctl -xe
mai 14 21:08:49 raspberrypi kernel: IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
mai 14 21:08:49 raspberrypi hostapd[10039]: nl80211: Could not configure driver mode
mai 14 21:08:49 raspberrypi hostapd[10039]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
mai 14 21:08:49 raspberrypi avahi-daemon[352]: Interface wlan1.IPv4 no longer relevant for mDNS.
mai 14 21:08:49 raspberrypi avahi-daemon[352]: Leaving mDNS multicast group on interface wlan1.IPv4 with address 192.168.10.1.
mai 14 21:08:50 raspberrypi hostapd[10039]: nl80211 driver initialization failed.
mai 14 21:08:50 raspberrypi hostapd[10039]: wlan1: interface state UNINITIALIZED->DISABLED
mai 14 21:08:50 raspberrypi hostapd[10039]: wlan1: AP-DISABLED
mai 14 21:08:50 raspberrypi hostapd[10039]: hostapd_free_hapd_data: Interface wlan1 wasn't started
mai 14 21:08:50 raspberrypi avahi-daemon[352]: Withdrawing address record for 192.168.10.1 on wlan1.
mai 14 21:08:50 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited status=1
mai 14 21:08:50 raspberrypi kernel: IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
mai 14 21:08:50 raspberrypi avahi-daemon[352]: Joining mDNS multicast group on interface wlan1.IPv4 with address 192.168.10.1.
mai 14 21:08:50 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
-- Subject: L'unité (unit) hostapd.service a échoué
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- L'unité (unit) hostapd.service a échoué, avec le résultat failed.
mai 14 21:08:50 raspberrypi avahi-daemon[352]: New relevant interface wlan1.IPv4 for mDNS.
mai 14 21:08:50 raspberrypi systemd[1]: hostapd.service: Unit entered failed state.
mai 14 21:08:50 raspberrypi avahi-daemon[352]: Registering new address record for 192.168.10.1 on wlan1.IPv4.
mai 14 21:08:50 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
lines 1509-1531/1531 (END)

```log of systemctl status hostapd.service
oot@raspberrypi:~# systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authent
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: 
   Active: activating (start) since Tue 2019-05-14 21:04:00 CEST; 189ms ago
Cntrl PID: 9863 (hostapd)
      CPU: 17ms
   CGroup: /system.slice/hostapd.service
           └─9863 /usr/sbin/hostapd -P /run/hostapd.pid -B /etc/hostapd/hostapd.

mai 14 21:04:00 raspberrypi systemd[1]: Starting Advanced IEEE 802.11 AP and IEE
mai 14 21:04:00 raspberrypi hostapd[9863]: Configuration file: /etc/hostapd/host
lines 1-10/10 (END)

I tried with 3 sd card, with the last version of raspian stretch and raspian stretch lite, and i have the same issue

pihomeserver, 2019-05-14:
Hello
Only Raspbian Stretch Lite is supported.
Why did you change wlan0 to wlan1 ? There is an issue due to that. I guess you use another Wifi controller ?

indian65, 2019-05-14:
On the raspberrry the wifi range area are poor, i have a big house with two floors, i buy a external dongle wifi with ap and with antenna, it's recognise by the raspberry, it's why i change the line wlan 0 by wlan 1 ( the name of the dongle).

pihomeserver, 2019-05-15:
Can you please provide the output of `ifconfig -a`, `iwconfig wlan1`
Do you know what should be the driver of the dongle for linux system ? Also where you able to use it as a normal interface with wpa supplicant configuration ?

indian65, 2019-05-15:
the log of ifconfig
pi@raspberrypi:~ $ ifconfig -a
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:a8:8e:e0:bc  txqueuelen 0  (Ethernet)
        RX packets 8959  bytes 828849 (809.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10960  bytes 14943342 (14.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.14  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::610d:a5fb:c513:54e  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8c:e6:dd  txqueuelen 1000  (Ethernet)
        RX packets 442349  bytes 554785116 (529.0 MiB)
        RX errors 0  dropped 252  overruns 0  frame 0
        TX packets 319462  bytes 47865921 (45.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 99  bytes 9408 (9.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 99  bytes 9408 (9.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:d9:b3:88  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  broadcast 192.168.10.255
        ether 00:c0:ca:97:90:ef  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
the log of wlan1
pi@raspberrypi:~ $ iwconfig wlan1
wlan1     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
The driver of this dongle is the same of the internal wifi of the raspberry

pihomeserver, 2019-05-15:
Seems to be purely wifi configuration (not the script). 
You should try googling the error message may be there? https://unix.stackexchange.com/questions/215030/hostapd-error-nl80211-could-not-configure-driver-mode

indian65, 2019-05-17:
It's work
I add after the fail
`killall wpa_supplicant`
 hostapd /etc/hostapd/hostapd.conf

my wifi dongle are enabled, and i run manually the last lines, one by one, of your script.
I have now the wifi with a large area.
Thanks for your help.

indian65, 2019-05-17:
I have a question, it is possible to select a user in the nfdump file for see only him in the history log? I have five users in my database ( i have a guest house with five bedrooms) .

pihomeserver, 2019-05-17:
I guess i should add the killall somewhere ...

User is set in freeradius, nfdump is packets tracking. I guess that could be more usefull to use information in the radacct table to extract statistics, MAC address that you could use to filter on nfdump output

indian65, 2019-05-25:
I have a new issue, after a reboot, if i have files in nfdump folder, but all the files are empty.
root@Condorinette-Wifi:~# nfdump -R /var/cache/nfdump/2019-05-25
Date first seen          Duration Proto      Src IP Addr:Port          Dst IP Addr:Port   Packets    Bytes Flows
No matched flows

pihomeserver, 2019-05-25:
@indian65 if it's a new issue please open a new ticket ;)

closed 2019-05-25 by pihomeserver